jopohl/urh #611: Filter file types in dir

lunakk-PL, 2019-01-18:
##### Expected Behavior
displaying previously saved files in protocol dir

##### Actual Behavior
list is empty
##### Steps To Reproduce

1. Go to 'Save protocol as'
2. when saving file type marked as "(*proto.xml *.proto)" is selected previously saved files in dir are invisible
3. after changing file type to "All files (*)" files are showing correctly in directory

##### Screenshots
[image]

##### Platform Specifications
OS: Windows 10 x64 ver. 10.0.17134
URH version: 2.5.4
Python version: 3.7.2
Installed via msi

andynoack, 2019-01-18:
The filter is for files ending with *.proto.xml or *.proto, __not__ only *.xml!

lunakk-PL, 2019-01-18:
I understand, thx.
BTW great software :)

closed 2019-01-18 by andynoack