jopohl/urh #615: ZeroDivisionError exception while open cfile

oripio, 2019-02-14:
##### Expected Behavior
While open cfile I'm getting error:

Error: float division by zero

----------

Traceback (most recent call last):
File "/usr/local/lib/python3.5/dist-packages/urh/controller/MainController.py", line 781, in show_open_dialog
self.add_files(file_names)
File "/usr/local/lib/python3.5/dist-packages/urh/controller/MainController.py", line 402, in add_files
self.add_signalfile(filename, group_id, enforce_sample_rate=enforce_sample_rate)
File "/usr/local/lib/python3.5/dist-packages/urh/controller/MainController.py", line 288, in add_signalfile
self.add_signal(signal, group_id)
File "/usr/local/lib/python3.5/dist-packages/urh/controller/MainController.py", line 302, in add_signal
signal.auto_detect(detect_modulation=True, detect_noise=False)
File "/usr/local/lib/python3.5/dist-packages/urh/signalprocessing/Signal.py", line 381, in auto_detect
estimated_params = AutoInterpretation.estimate(self.data, **kwargs)
File "/usr/local/lib/python3.5/dist-packages/urh/ainterpretation/AutoInterpretation.py", line 378, in estimate
center = detect_center(msg_rect_data)
File "/usr/local/lib/python3.5/dist-packages/urh/ainterpretation/AutoInterpretation.py", line 207, in detect_center
y, x = np.histogram(rect, bins=np.arange(hist_min, hist_max + hist_step, hist_step))
ZeroDivisionError: float division by zero

##### Actual Behavior

##### Steps To Reproduce

1. Go to 'File'
2. Click on 'Open'
3. Select any cfile
4. Click on 'Open'
5. Get an error

##### Screenshots

##### Platform Specifications

 - OS: Debian Linux (Linux Mint LMDE 3 Cindy)
 - URH version: 2.5.5
 - Python version: 3.5.3
 - Installed via pip

jopohl, 2019-02-14:
Can't fix this right now as I am on vacation but as a workaround you can use Edit menu and uncheck the auto detect newly loaded signals (exact caption may vary) checkbox.

jopohl, 2019-02-14:
Furthermore it would be cool if you can upload the signal which causes this crash for further investigation.

commit 8ea7feb9 (2019-02-20): fix center detection crash on special signals (fix #615)

If a signal has segments with zero variance i.e. constant lines,
the center detection crashes because the histogram can not be created.
This fix catches the ZeroDivision error and returns None in that case.
closed 2019-02-20 by jopohl