jopohl/urh #619: Edit -> Decoding->External Program-> Decoder, Encoder File Chooser grayed out.

Jackfritt, 2019-02-20:
##### Expected Behavior
Encoder and Decoder scripts should be chooseable

##### Actual Behavior
Button "..." for Decoder and Decoder are greyed out. Nothing happpens on Mouseclick

##### Steps To Reproduce

1. Go to 'Edit'
2. Click on 'Decoding'
3. Choose 'External Program'
4. Try to click on Decoder "..." Button or Encoder "..." Button both are greyed out.

##### Platform Specifications

 - OS: [Debian 9.7]
 - URH version: [e.g. 2.5.5]
 - Python version: [e.g. 3.5]
 - Installed via [e.g. pip]

jopohl, 2019-02-21:
Did you drag and drop the "External Program" primitive into the decoding chain?

[image]

Jackfritt, 2019-02-21:
Thx for that hint. Workes like a charm. Fastest support ever.

closed 2019-02-21 by Jackfritt