jopohl/urh #623: Failing tests

FFY00, 2019-02-25:
```
=================================== FAILURES ===================================
_______________________ TestFilter.test_bandpass_filter ________________________
:-1: running the test CRASHED with signal 11
_______________________ TestFilter.test_fft_convolution ________________________
:-1: running the test CRASHED with signal 11
________________________ TestFilter.test_filter_caption ________________________
:-1: running the test CRASHED with signal 11
______________________ TestFilter.test_filter_full_signal ______________________
:-1: running the test CRASHED with signal 11
_______________________ TestFilter.test_filter_selection _______________________
:-1: running the test CRASHED with signal 11
__________________________ TestFilter.test_fir_filter __________________________
:-1: running the test CRASHED with signal 11
_____________________ TestPlugins.test_insert_sine_plugin ______________________
:-1: running the test CRASHED with signal 11
____________________ TestPlugins.test_message_break_plugin _____________________
:-1: running the test CRASHED with signal 11
____________________ TestPlugins.test_sdr_interface_plugin _____________________
:-1: running the test CRASHED with signal 11
__________________ TestPlugins.test_zero_hide_plugin_function __________________
:-1: running the test CRASHED with signal 11
____________________ TestPlugins.test_zero_hide_plugin_gui _____________________
:-1: running the test CRASHED with signal 11
=============================== warnings summary ===============================
src/urh/plugins/InsertSine/InsertSinePlugin.py:49
  /build/urh/src/urh-2.5.6/src/urh/plugins/InsertSine/InsertSinePlugin.py:49: DeprecationWarning: invalid escape sequence \.
    QRegExpValidator(QRegExp("[0-9]+([nmµ]*|([\.,][0-9]{1,3}[nmµ]*))?$")))

src/urh/simulator/Simulator.py:202
  /build/urh/src/urh-2.5.6/src/urh/simulator/Simulator.py:202: DeprecationWarning: invalid escape sequence \.
    regex = "(item[0-9]+\.counter_value)"
```

Build routine in https://git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/urh.

jopohl, 2019-02-26:
I noticed some failing tests in my CI too. It is caused  by PyQt5 5.12 (prior versions work without problem). So workaround is to use PyQt != 5.12 to get all tests passing.

FFY00, 2019-02-26:
Yeah, it's not really an option. Do you know how this could be fixed? Otherwise I can just disable those tests but that might still break stuff.

jopohl, 2019-02-26:
It should be safe to disable those tests, because the error is related to some cleanup stuff in the Test routine (just verified that).

FFY00, 2019-02-26:
Thanks for confirming that. This still should be fixed though.

jopohl, 2019-02-26:
I agree, but from experience I can say that this will most likely will be fixed with time by the PyQt guys. I had a similar behaviour on Windows with the last version of PyQt5, you can see it from the  ``` data/requirements.txt  ``` :

```
...
pyqt5; sys_platform != 'win32' and sys_platform != 'linux'
pyqt5!=5.11.1,!=5.11.2,!=5.11.3; sys_platform == 'win32'
pyqt5!=5.12; sys_platform == 'linux'
...
```

where I skipped the versions 5.11.1-3 for Windows, but version 5.12 is working fine there again. So right now I see few reasons to do anything about this from URH side. if you want to speed things up you may get in contact with the PyQt developers.

closed 2019-02-26 by jopohl