jopohl/urh #642: Fuzzing using continuous send mode does not work

mdeeg, 2019-04-15:
##### Expected Behavior
Send data packets shown in _Generated Data_ table.

##### Actual Behavior
Show error message regarding missing device (see screenshot) and do not send any data.

##### Steps To Reproduce
1. Load URH project
2. Click on _Generator_ tab
3. Drag defined protocol to _Generated Data_ table
4. Click on _Fuzzing_ tab
5. Edit fuzzing label for packet data to fuzz
6. Add many fuzzed values (e.g. more than 1000)
7. Close edit fuzzing dialog
8. Checkmark fuzzing label
9. Click _Fuzz_ button
10. Click _Send data..._ button
11. Click OK button of warning message "Entering continuous send mode"
12. Error message concerning missing device (see screenshot)

##### Screenshots
[image]

##### Platform Specifications
 - OS: Arch Linux
 - URH version: 2.5.7
 - Python version: 3.7.3
 - Installed via pip
 - SDR: HackRF One

jopohl, 2019-04-15:
Continuous send mode only works with devices that have __native backend__ configured. So make sure you have at least one SDR with native backend in Options -> Device.

mdeeg, 2019-04-15:
Hello Johannes,
thanks, for the quick reponse. By selecting the **native backend** for the HackRF One, no error message is shown and packets are sent.

closed 2019-04-15 by mdeeg