jopohl/urh #647: Time changes for interpretation view

hanskrst, 2019-05-02:
##### Expected Behavior

When selecting part of capture in Interpretation view I expect the time shown for the highlighted signal to be adjusted for sampling rate in such a way that it represents actual time.

##### Actual Behavior
The same signal captured with sampling rate og 2MSps is shown to last for 682.34ms while a sampling rate of 20MSps gets reported as 6.82s.
Capturing with 10MSps gets reported as ~1.36s, while 5MSps gets reported as 682.56ms (same as for 2 MSps)

##### Steps To Reproduce

Perform capture with low sampling rate, recapture with high samplingrate. Open both captures in Interpretation view. Highlight similar parts of captures. 

##### Screenshots

[image]

##### Platform Specifications

 - OS: Win10 v.10.0.17134 
 - URH version: 2.5.7

jopohl, 2019-05-03:
The sample rate can be configured in the Info Dialog of the signal which is accessible with the info button right next to the close button.

hanskrst, 2019-05-03:
I see, now I see it is also documented in the userguide. 
Sorry for the bother, and thanks for pointing it out!

closed 2019-05-03 by hanskrst