jopohl/urh #648: Interpretation displaying inphase and quadrature, FM, PM and AM

ComputerCarsten, 2019-05-03:
The interpretation tab is only showing one component of the analog signal however SDR tools produce inphase and quadrature components aka IQ-Data which is treated as complex numbers.

I would've prefered a multi color oscillogram which shows I and Q and maybe 
AM: sqrt(I(z)^2 + Q(z)^2)
FM: unwrap(atan2(I(z), Q(z)) - atan2(I(z-1), Q(z-1)))
PM: unwrap(atan2(I(z), Q(z))
at the same time. 
FM is highly sampling-rate dependend (among other things) so a gain-slider is needed.
PM normally gives a messy output (drifts to infinity, or wrapping at +/- PI when not unwrapped) when there is a carrier offset. So this should be optional in any case.

Don't forget the graph legend and scales.

jopohl, 2019-05-03:
Have you switched to the ``` Demodulated ``` view? Basically for each modulation you will see a rectangular curve based on the formulas you given for each modulation type.

If you mean something different, a screenshot/picture would be helpful.

ComputerCarsten, 2019-05-03:
At the very end of the video you can see of how i like to see the graph composed. Scales and a legend would be nice also. Video is in German.
https://www.youtube.com/watch?time_continue=675&v=974JVSnXXKo

jopohl, 2019-05-03:
The different views you describe are already implemented in URH. Here is an example for FM:
[image]
Simply switch to the demodulated view and you get the instantaneous frequencies. The same goes for amplitude/phase, when you set the modulation type to ASK/PSK.

I do not see a great benefit in merging all these views into one plot, especially since the modulation type is automatically detected anyway.

jopohl, 2019-05-08:
Closing due to inactivity.

closed 2019-05-08 by jopohl