jopohl/urh #650: libEGL WARNING DRI2 failed to authenticate

MACH1SCJ, 2019-05-06:
Getting this error at start up.

pi@raspberrypi:~ $ urh
libEGL warning: DRI2: failed to authenticate
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
inotify_add_watch("/home/pi/.config/qt5ct") failed: "No such file or directory"
[INFO::SettingsProxy.py::get_receive_buffer_size] Allocate receive buffer with 262.144kB
[INFO::Device.py::start_rx_mode] RTLSDR: Starting RX Mode
Found Rafael Micro R820T tuner
[INFO::Device.py::log_retcode] RTLSDR-OPEN (#0): Success

##### Platform Specifications
 - Rpi3
 - NESDR SMART RTL-SDR
 -OS- rpi image
 - URH version: - 2.5.7
 - Python version:  - 3.5.3 ( and 2.7.13 is default (GCC 6.3.0 20170516) on linux2 )
 - Installed via - "sudo pip3 install ." from within urh download  -- pip and pip3 both report V. 19.1 from python 3.5

jopohl, 2019-05-06:
Is the warning affecting you in any way? It seems to be working fine with RX from what I see in the log.

The libEGL warning seems to be a problem with Qt on Pi, nothing specific to URH. You can find a possible solution [here](https://raspberrypi.stackexchange.com/questions/61078/qt-applications-dont-work-due-to-libegl).

jopohl, 2019-05-08:
Closing due to inactivity.

closed 2019-05-08 by jopohl