jopohl/urh #659: PlutoSDR - when transmittion is started the first thing transmitted is some junk

Matjaz-B, 2019-06-10:
##### Expected Behavior
Only actual signals should be transmitted

##### Actual Behavior
When starting transmission the first thing transmitted is some highpower noise. After that the actual data is transmitted. It seems like FIFO is not cleared and some obsolete data is transmitted before actual data.

##### Steps To Reproduce
1. Open two URHs (or spectrum analyser)
2. Use one URH as receiver
3. Use one URH as transmitter - set transmitt buffer to all muted data
4. Start transmission for 1s and stop if afterwards - repeat for few times
5. On reception side you will see received peaks

Or - transmit data via command line:
`urh_cli.exe -d PlutoSDR -f 868000000.0 -s 3e6 -bb -20 -g -70 -cf 868000000.0 -mo FSK -bl 61 -p0=-50000.0 -p1=+50000.0 -m 555559999999999999999999999999999/10ms  -tx --hex`
The issue is more visible, when gain is set to low value. The initial _noise_  is quite wideband and amplitude is almost at 0 dBm.

##### Screenshots
[image]

##### Platform Specifications
 - OS: Windows 10
 - URH version: 2.7.0 (the same behaviour was in older versions)
 - Python version: The one embedded in setup file.
 - Installed via setup file - Universal.Radio.Hacker-2.7.0-x64.exe

andynoack, 2019-06-10:
Well, that is a behaviour we can observe with many SDRs when they change into transmit mode and we think that it is a hardware or driver problem. Can you test with another TX software (e.g. Gnuradio), if the problem occurs there too?

andynoack, 2019-06-15:
Any news here? Can we close the Issue?

Matjaz-B, 2019-06-17:
Thanks for response.
I did not know that this is genral problem (SW and HW). Frankly before this use case I was always using higher output power (more than -50 dBm) where this behaviour was not important/detected.
I will try to reproduce this issue with GNU Radio to locate the source of the problem. I hope that I'll something to show by end of this week.

Thanks

vsboost, 2019-06-26:
I get the same with LimeSDR-mini - huge spike of junk at startup, prior to the signal being replayed even starts, also notice once i hit stop the mini still is in TX, have to go to receive a signal for it to stop the TX.

jopohl, 2019-06-29:
Just verified with a very simple GNU Radio Flowgraph that the behaviour is also present there:
[image]

Every time I run the flow graph I record a short signal as described in this issue.

Therefore, I assume this is a hardware and/or driver problem of PlutoSDR and we cannot do much about it at URH side. You might try reporting this at [the LimeSuite repo](https://github.com/myriadrf/LimeSuite).

closed 2019-06-29 by jopohl

jopohl, 2019-06-30:
@vsboost just saw that this is an open issue with LimeSDR-mini: https://github.com/myriadrf/LimeSuite/issues/235

vsboost, 2019-07-01:
Thanks Johannes, I was unaware this is an issue with the mini itself..