jopohl/urh #667: 2.7.0 64-bit Windows release spectrogram weirdness

klohner, 2019-06-25:
##### Expected Behavior
In spectrogram view, single frequency of RTL_SDR signal is expected to be shown in proper frequency row.  This is the behavior in 2.6.0 release, but not in 2.7.0 release.

##### Actual Behavior
When Windows 64-bit URH 2.7.0 captures are displayed as spectrogram, single frequency of seems to be displayed at multiple rows. Selecting a signal row here and applying bandpass filter results seem to be of incorrect signal frequency and spectrogram display is also shown incorrectly.

##### Steps To Reproduce
Capture signal in Windows 64-bit 2.7.0 release using SDR_RTL device, save, and change view to Spectrogram.  Behavior is also observed when loading previous saved samples, even from previous versions (i.e. 2.6.0) of URH.

##### Screenshots
All screenshots are from an ASK signal from a LaCrosse temperature sensor using a SDR_RTL (NooElec NESDR SMArt).

[image]
Expected behavior, as seen in 2.6.0.  Top view is recorded sample spectrogram, signal frequency selected. Bottom view is spectrogram view after bandpass applied.

---

[image]
Top view is the same sample as recorded in 2.6.0, loaded and displayed in 2.7.0.  Signal seems to appear in 4 different frequencies and length of sample file seems longer any maybe truncated? Bottom view is spectrogram view after bandpass applied to bottom signal frequency. It seems like it did not actually pass the selected signal.

---

[image]
This is essentially the same capture and display process as originally performed in 2.6.0 as performed in 2.7.0.  Sample looks great in Analog view, but this spectrogram view looks very different that what is expected and performs strangely.

---

[image]
This is 2.7.0 showing the analog view of that 2.7.0-recorded signal. Quality seems much better than what was captured in 2.6.0, but spectrogram view is not functioning properly.

---

[image]
This is the sample recorded in 2.7.0 as loaded and displayed in 2.6.0 spectogram view.  This seems weird too as the single-frequency signal seems duplicated in several rows.

---

[image]

This is 2.7.0 showing analog view of the signal as captured in 2.7.0 (top) vs the signal as recorded and saved in 2.6.0 and loaded into 2.7.0.  Both views are zoomed in a bit in time and amplitude. I like how 2.7.0 seems to have little to no noise before signal starts, and it seems to even pick up a signal level slightly above center for the GAPs in the signal. Is this an artifact of the new URH signal capture, or is the device actually sending GAPs as a lower ASK level?

##### Sample Files as recorded in 2.6.0 and 2.7.0

These are the samples as captured and saved in each 2.6.0 and 2.7.0.  Method for generating and capturing each signal was the same in each version.

[URH_2.6.0_and_2.7.0_samples.zip](https://github.com/jopohl/urh/files/3326683/URH_2.6.0_and_2.7.0_samples.zip)

##### Platform Specifications

 - OS: Windows 64-bit
 - URH version: 2.7.0
 - Installed via 64-bit Release .exe file.

jopohl, 2019-06-25:
Just ran a quick test with a RTL-SDR (DVB-T stick) and could not reproduce that error. Of course I have not your signal source available. You might try unchecking the DC correction:
[image]

This should bring you more close to the setting of v2.6.0.

jopohl, 2019-06-25:
If this not helps, could you make a capture with v2.6.0 and export it as complex16s? This way I can see if the error is in recording or calculation of spectrogram for complex16 files.

closed 2019-06-25 by jopohl

klohner, 2019-06-25:
[image]
This is how I sampled the signal in 2.7.0.

---

[image]
And here is that signal as sampled and viewed in 2.7.0 as Analog, Demodulated, and Spectrogram.  Spectrogram view is still weird.  This is an ASK signal, but it appears at 4 frequencies, and the display is truncated.

---

[image]
Here is how I sampled the signal in 2.6.0.  I'm using the same settings.  I saved it as a complex16s.

---

[image]
Here is that signal as sampled and viewed in 2.6.0 as Analog, Demodulated, and Spectrogram.  This is expected Spectrogram.

---

Here are the sample files as complex16s.
[URH_2.6.0_and_2.7.0_samples (2).zip](https://github.com/jopohl/urh/files/3326932/URH_2.6.0_and_2.7.0_samples.2.zip)

I've observed this behavior on two separate Windows 10 machines using the 64-bit installation release files.

jopohl, 2019-06-25:
Just verified that the fix is working with the two signals you provided. So you can either run from source (fix is merged) or wait till next release.

klohner, 2019-06-25:
Thank you. I really love this program and look forward to new releases.  I'm impressed you resolved this issue in less time than it took me to write it up!

I can confirm that it's now displaying spectrograms properly from saved files.  Unfortunately I can't figure out how to run from source with support for rtl-sdr enabled to create new samples. Is there a document for this? If not, I'm happy to wait until the next release.  Thanks!

jopohl, 2019-06-26:
On windows this is indeed a bit tricky. But I just made a new release so feel free to use that. :+1:

commit d33861d4 (2019-06-25): fix #667