jopohl/urh #670: LimeSDR: Failed to receive stream

derlucas, 2019-06-28:
##### Expected Behavior
run URH and capture data or use spectrum analyzer to record data or show spectrum.

##### Actual Behavior
no data is recorded and nothing showing up in spectrum analyzer.

##### Steps To Reproduce
install LimeSuite and URH
Open URH, click on File->Record Signal
Select the LimeSRD Device
Click on "Start"
Nothing happens, console spits out following:

```
[INFO::SettingsProxy.py::get_receive_buffer_size] Allocate receive buffer with 8,971GB
[INFO::Device.py::start_rx_mode] LimeSDR: Starting RX Mode
RemoteControl Listening on port: 5000

Reference clock 30.72 MHz
[INFO::Device.py::log_retcode] LimeSDR-OPEN (LimeSDR-USB, media=USB 3.0, module=FX3, addr=1d50:6108, serial=00****): Success
[INFO::Device.py::log_retcode] LimeSDR-INIT: Success
[INFO::Device.py::log_retcode] LimeSDR-SET_CHANNEL_INDEX to 0: Success
[INFO::Device.py::log_retcode] LimeSDR-SET_ANTENNA_INDEX to 2: Success
[INFO::Device.py::log_retcode] LimeSDR-SET_FREQUENCY to 433920000.0: Success
[INFO::Device.py::log_retcode] LimeSDR-SET_SAMPLE_RATE to 2000000.0: Success
RX LPF configured
[INFO::Device.py::log_retcode] LimeSDR-SET_BANDWIDTH to 2000000.0: Success
[INFO::Device.py::log_retcode] LimeSDR-SET_RF_GAIN to 0.25: Success
[INFO::Device.py::log_retcode] LimeSDR-Initialize stream: Success
[WARNING::LimeSDR.py::receive_sync] LimeSDR: Failed to receive stream
[WARNING::LimeSDR.py::receive_sync] LimeSDR: Failed to receive stream
[WARNING::LimeSDR.py::receive_sync] LimeSDR: Failed to receive stream
....
[INFO::Device.py::stop_rx_mode] LimeSDR: Stopping RX Mode: Stopped receiving: Stop button clicked
[WARNING::LimeSDR.py::receive_sync] LimeSDR: Failed to receive stream
[INFO::Device.py::log_retcode] LimeSDR-CLOSE: Success
[INFO::Device.py::read_receiving_queue] EOF Error: Ending receive thread
[DEBUG::Device.py::read_receiving_queue] Exiting read_receive_queue thread.
[INFO::Device.py::read_device_messages] Exiting read device message thread due to 
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread
[DEBUG::Device.py::stop_rx_mode] Closing parent control connection: handle is closed
[INFO::Device.py::stop_rx_mode] LimeSDR: Stopping RX Mode: Dialog closed. Killing recording process.
[DEBUG::SendRecvDialog.py::closeEvent] Device stopped successfully.
[DEBUG::SendRecvDialog.py::closeEvent] Cleaning up device
[DEBUG::SendRecvDialog.py::closeEvent] Successfully cleaned up device
```

##### Platform Specifications
 - OS: Arch Linux
 - URH version: 2.7.2
 - LimeSuite version: 19.01.0-1
 - Python version: 3.7.3
 - Installed via pacman

jopohl, 2019-06-28:
Can't reproduce. I develop on Arch Linux :wink: My LimeSDR works fine here. 

* Are you able to use your LimeSDR in other software e.g. LimeSuiteGUI itself?
* Is your firmware up to date? Posting the output of ``` LimeQuickTest ``` might help.

jopohl, 2019-07-01:
Closing due to inactivity.

derlucas, 2019-07-01:
Yes LimeQuickTest is working.

```[ TESTING STARTED ]
->Start time: Mon Jul  1 22:33:19 2019
->Device: LimeSDR-USB, media=USB 3.0, module=FX3, addr=1d50:6108, serial=xx
  Serial Number: xx

[ Clock Network Test ]
->FX3 GPIF clock test
  Test results: 27049; 30805; 34561 - PASSED
->Si5351C test
  CLK0: 17554 / 17554 - PASSED
  CLK1: 17554 / 17554 - PASSED
  CLK2: 17554 / 17554 - PASSED
  CLK3: 17554 / 17554 - PASSED
  CLK4: 17554 / 17554 - PASSED
  CLK5: 17554 / 17554 - PASSED
  CLK6: 17554 / 17554 - PASSED
->ADF4002 Test
  Result: 10 - PASSED
->VCTCXO test
  Results : 5112950 (min); 5113092 (max) - PASSED
->Clock Network Test PASSED

[ FPGA EEPROM Test ]
->Read EEPROM
->Read data: 13 02 17 13 02 17 02
->FPGA EEPROM Test PASSED

[ LMS7002M Test ]
->Perform Registers Test
->External Reset line test
  Reg 0x20: Write value 0xFFFD, Read value 0xFFFD
  Reg 0x20: value after reset 0x0FFFF
->LMS7002M Test PASSED

[ RF Loopback Test ]
Note: The test should be run without anything connected to RF ports
->Configure LMS
->Run Tests (TX_2-> LNA_L):
  CH0 (SXR=800.0MHz, SXT=805.0MHz): Result:(-14.1 dBFS, 5.00 MHz) - PASSED
  CH1 (SXR=800.0MHz, SXT=805.0MHz): Result:(-15.7 dBFS, 5.00 MHz) - PASSED
->Run Tests (TX_1 -> LNA_W):
  CH0 (SXR=1800.0MHz, SXT=1805.0MHz): Result:(-15.3 dBFS, 5.00 MHz) - PASSED
  CH1 (SXR=1800.0MHz, SXT=1805.0MHz): Result:(-17.0 dBFS, 5.00 MHz) - PASSED
->Run Tests (TX_2-> LNA_H):
  CH0 (SXR=2500.0MHz, SXT=2505.0MHz): Result:(-14.9 dBFS, 5.00 MHz) - PASSED
  CH1 (SXR=2500.0MHz, SXT=2505.0MHz): Result:(-14.6 dBFS, 5.00 MHz) - PASSED
->RF Loopback Test PASSED

=> Board tests PASSED <=

Elapsed time: 1.85 seconds```

Using it with GnuRadio is fine, so is qspectrumanalyser.

jopohl, 2019-07-01:
Same behaviour when you run URH from source?

derlucas, 2019-07-01:
It has the same behavior the first time i used it from source.
Now i checked out the github repo again and it runs fine. Maybe it works now because i installed LimeUtils from source also.

closed 2019-07-01 by derlucas