jopohl/urh #676: LimeSDR Mini: Failed to install URH

furang, 2019-07-04:
##### Expected Behavior
URH installs without any problems

##### Actual Behavior
URH not installed with error

 ERROR: Complete output from command python setup.py egg_info:
    ERROR: Skipping native support for usrp
    Skipping native support for rtlsdr
    Skipping native support for plutosdr
    Skipping native support for sdrplay
    Skipping native support for hackrf
    Found LimeSuite lib. Will compile with native limesdr support
    Skipping native support for bladerf
    Skipping native support for airspy
    
    Error compiling Cython file:
    ------------------------------------------------------------
    ...
        :return: 0 on success, (-1) on failure
        """
        stream.isTx = IS_TX
        stream.channel = <uint32_t> CHANNEL
        stream.fifoSize = fifo_size
        stream.dataFmt = dataFmt_t.LMS_FMT_F32
                                 ^
    ------------------------------------------------------------
    
    src/urh/dev/native/lib/limesdr.pyx:356:30: 'dataFmt_t' is not a constant, variable or function identifier
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-g6v4gvlh/urh/setup.py", line 145, in <module>
        ext_modules=get_extensions(),
      File "/tmp/pip-install-g6v4gvlh/urh/setup.py", line 104, in get_extensions
        extensions = cythonize(extensions, compiler_directives=COMPILER_DIRECTIVES, quiet=True, compile_time_env=device_extras)
      File "/usr/lib/python3/dist-packages/Cython/Build/Dependencies.py", line 877, in cythonize
        cythonize_one(*args)
      File "/usr/lib/python3/dist-packages/Cython/Build/Dependencies.py", line 997, in cythonize_one
        raise CompileError(None, pyx_file)
    Cython.Compiler.Errors.CompileError: src/urh/dev/native/lib/limesdr.pyx

##### Steps To Reproduce

1. pip install urh
2. Wait for error to raise

##### Screenshots

##### Platform Specifications

 - OS: Ubuntu 16.04.6 LTS x64
 - URH version: ???
 - Python version: Python 3.5.2 (default, Nov 12 2018, 13:43:14), also tried on Python 2.7.12 (default, Nov 12 2018, 14:36:49)
 - Cython version 0.23.4

 - Installed via pip, also tried pip3

LimeUtil --info
######################################################
## LimeSuite information summary
######################################################

Version information:
  Library version:	v19.04.0-myriadrf1~xenial
  Build timestamp:	2019-05-13
  Interface version:	v2019.1.0
  Binary interface:	19.04-1

System resources:
  Installation root:	/usr
  User home directory:	/home/furang
  App data directory:	/home/furang/.local/share/LimeSuite
  Config directory:	/home/furang/.limesuite
  Image search paths:
     - /home/furang/.local/share/LimeSuite/images
     - /usr/share/LimeSuite/images

Supported connections:
   * FT601
   * FX3
   * PCIEXillybus

LimeSUite installed using oficial guire https://wiki.myriadrf.org/Installing_Lime_Suite_on_Linux (Installing via PPA)

jopohl, 2019-07-05:
Verified this in a docker container. Problem is that Ubuntu 16.04 has a broken cython version in the repos. You can fix it like this:

```
apt-get remove cython3
pip3 install cython
pip3 install urh
```

furang, 2019-07-05:
Thanks a lot. URH now builds with no problem.
Your option works like a charm.
Issue is resolved.

closed 2019-07-05 by furang