jopohl/urh #679: MacOS DMG installer doesn't start

GeckoGB, 2019-08-01:
Hi,
Whenever i try to run the installer from DMG package (version 2.7.0 through 2.7.3), the application doesn't start (icon is jumping a few seconds, then disappears).

Messages in console:
```
Aug  1 11:08:58 computer com.apple.xpc.launchd[1] (com.apple.TMHelperAgent.SetupOffer): Service only ran for 2 seconds. Pushing respawn out by 8 seconds.
Aug  1 11:08:58 computer com.apple.xpc.launchd[1] (com.apple.auditd[7507]): Endpoint has been activated through legacy launch(3) APIs. Please switch to XPC or bootstrap_check_in(): com.apple.auditd
Aug  1 11:08:58 computer com.apple.xpc.launchd[1] (com.apple.imfoundation.IMRemoteURLConnectionAgent): Unknown key for integer: _DirtyJetsamMemoryLimit
Aug  1 11:08:59 --- last message repeated 1 time ---
Aug  1 11:08:59 computer com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.pid.IDECacheDeleteAppExtension.7504): Path not allowed in target domain: type = pid, path = /Applications/Xcode.app/Contents/Frameworks/IDEKit.framework/Versions/A/XPCServices/IDETouchBarSimulatorService.xpc error = 147: The specified service did not ship in the requestor's bundle, origin = /Applications/Xcode.app/Contents/PlugIns/IDECacheDeleteAppExtension.appex
Aug  1 11:08:59 computer com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.pid.IDECacheDeleteAppExtension.7504): Path not allowed in target domain: type = pid, path = /Applications/Xcode.app/Contents/SharedFrameworks/LLDB.framework/Versions/A/XPCServices/RootDebuggingXPCService.xpc error = 147: The specified service did not ship in the requestor's bundle, origin = /Applications/Xcode.app/Contents/PlugIns/IDECacheDeleteAppExtension.appex
Aug  1 11:09:02 computer com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x1000003b.urh[7503]): Service exited due to signal: Killed: 9 sent by CoreServicesUIAgent[6544]
Aug  1 11:09:04 computer com.apple.xpc.launchd[1] (com.apple.WebKit.Networking.BEDCA85D-32F7-44A8-89E5-A3F42EAB4D51[7517]): Service exited with abnormal code: 1
Aug  1 11:09:04 computer com.apple.xpc.launchd[1] (com.apple.WebKit.Networking.5DF646F8-8AFA-4B35-BF76-BD66B7B1C807[7518]): Service exited with abnormal code: 1
Aug  1 11:09:04 computer com.apple.xpc.launchd[1] (com.apple.WebKit.Networking.3DE03845-28CF-4C5A-98AD-53C17F03BC56[7519]): Service exited with abnormal code: 1
Aug  1 11:09:05 computer com.apple.xpc.launchd[1] (com.apple.WebKit.Networking.7AB72ADA-C7D9-4006-94C8-B3635CD7AC41[7520]): Service exited with abnormal code: 1
Aug  1 11:09:09 computer com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x1000003c.urh[7515]): Service exited with abnormal code: 255
```

Also, if i try "show package content" and run "urh" executable in MacOS folder, i get a Terminal windows with this:
```
/Volumes/Universal.Radio.Hacker/urh.app/Contents/MacOS/urh ; exit;
[7550] Error loading Python lib '/var/folders/bz/msy8swb94h7132l9v1tjqsf40000gn/T/_MEImN4EtU/Python': dlopen: dlopen(/var/folders/bz/msy8swb94h7132l9v1tjqsf40000gn/T/_MEImN4EtU/Python, 10): Symbol not found: _futimens
  Referenced from: /var/folders/bz/msy8swb94h7132l9v1tjqsf40000gn/T/_MEImN4EtU/Python (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib
 in /var/folders/bz/msy8swb94h7132l9v1tjqsf40000gn/T/_MEImN4EtU/Python
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
Deleting expired sessions...none found.
```
MacOS 10.12.6, Macbook pro 15" mid-2015.

jopohl, 2019-08-01:
> MacOS 10.12.6, Macbook pro 15" mid-2015.

Guess this is too old. I build the DMG with OSX 10.14. Can you install URH with brew?

GeckoGB, 2019-08-01:
Thanks for your quick reply. Unfortunately i use macport, for which urh package doesn't seem to be available. That's why i was happy to see a DMG available!
Well, i should consider to updgrade my macOS someday...

jopohl, 2019-08-01:
Yeah updating may be an option :wink: 

For now, you can also try to install URH with pip:

```
python3 -m pip install urh
```

GeckoGB, 2019-08-01:
I was reluctant to try python install, because from my previous experience it is usually a mess between default macOS python 2.7 and additionnal python3, pip must be installed...
But i tried anyway, and i finally got it installed. It seems to work, but i have a lot of graphical bugs. Will have to test later on.
Thanks for your assistance and your great work!
Cheers.

jopohl, 2019-08-01:
You're welcome! I will close here. Feel free to open a new issue if you need further assistance e.g. with the graphical bugs (which may be rather Qt than URH itself related).

closed 2019-08-01 by jopohl