jopohl/urh #681: Pause threshold isn't respected in Interpretation window

sshawuk, 2019-08-14:
##### Expected Behavior
[image]

With a pause threshold of 8 bits, I expect the signal displayed in the above screenshot to be split at the 8 zero bits highlighted (amongst other repetitions of 8 zero bits).

##### Actual Behavior

The messages run together into one long message, ignoring the expected 8 bit pauses. 

##### Steps To Reproduce
1. Load a file for analysis. 
2. Automatically detect, and adjust parameters. 
3. Confirm that for this ASK signal, a pause threshold of 8 bits is set. 
4. Confirm that the signal does include 8 consecutive zero bits 

##### Platform Specifications

 - OS: Ubuntu
 - URH version: 2.7.3
 - Python version: 3.6.8
 - Installed via pip

jopohl, 2019-08-14:
Did you try 7?

sshawuk, 2019-08-14:
I did. It doesn't break this signal apart no matter how low I drop the threshold.

jopohl, 2019-08-14:
Can you copy your bits to a .txt file and upload it here so I can have a look?

sshawuk, 2019-08-14:
Sure thing, here you go :)

[sample_no_pause_detection.zip](https://github.com/jopohl/urh/files/3502931/sample_no_pause_detection.zip)

p.s. Really appreciate your attention. URH is a fantastic program!

jopohl, 2019-08-14:
Just had a look at your signal. Pause threshold refers to pause bits only (=bits coming from ranges in noise area). So an easy fix for your signal is to set the Noise threshold to "0.1412" and Pause Threshold to 7:

[image]

sshawuk, 2019-08-14:
Thank you!

I totally misunderstood exactly what constituted a pause bit! Really appreciate your help! Keep up the amazing work!

closed 2019-08-14 by sshawuk