jopohl/urh #683: LimeSDR  does not work // urh 2.7.4 win 3264

netdoggy, 2019-08-20:
```
Error: EnvironmentError: OSError: LoadLibrary failed to load "C:\sdr\PothosSDR\lib\uhd\modules\soapySupport.dll"
[INFO::SettingsProxy.py::get_receive_buffer_size] Allocate receive buffer with 262,144kB
[INFO::Device.py::start_rx_mode] LimeSDR: Starting RX Mode
Gateware version mismatch!
  Expected gateware version 2, revision 20
  But found version 2, revision 21
  Follow the FW and FPGA upgrade instructions:
  http://wiki.myriadrf.org/Lime_Suite#Flashing_images
  Or run update on the command line: LimeUtil --update

Reference clock 30,72 MHz
[INFO::Device.py::log_retcode] LimeSDR-OPEN: Success
[INFO::Device.py::log_retcode] LimeSDR-INIT: Success
[INFO::Device.py::log_retcode] LimeSDR-SET_CHANNEL_INDEX to 1: Success
[INFO::Device.py::log_retcode] LimeSDR-SET_ANTENNA_INDEX to 2: Success
[INFO::Device.py::log_retcode] LimeSDR-SET_FREQUENCY to 433920000.0: Success
[INFO::Device.py::log_retcode] LimeSDR-SET_SAMPLE_RATE to 2000000.0: Success
RX LPF configured
[INFO::Device.py::log_retcode] LimeSDR-SET_BANDWIDTH to 2000000.0: Success
[INFO::Device.py::log_retcode] LimeSDR-SET_RF_GAIN to 0.25: Success
[INFO::Device.py::log_retcode] LimeSDR-Initialize stream: Success
[WARNING::LimeSDR.py::receive_sync] LimeSDR: Failed to receive stream
[WARNING::LimeSDR.py::receive_sync] LimeSDR: Failed to receive stream
[WARNING::LimeSDR.py::receive_sync] LimeSDR: Failed to receive stream
[WARNING::LimeSDR.py::receive_sync] LimeSDR: Failed to receive stream
[WARNING::LimeSDR.py::receive_sync] LimeSDR: Failed to receive stream
...
...
...
[WARNING::LimeSDR.py::receive_sync] LimeSDR: Failed to receive stream
[WARNING::LimeSDR.py::receive_sync] LimeSDR: Failed to receive stream
[WARNING::LimeSDR.py::receive_sync] LimeSDR: Failed to receive stream
[INFO::Device.py::stop_rx_mode] LimeSDR: Stopping RX Mode: Stopped receiving: Stop button clicked
[WARNING::LimeSDR.py::receive_sync] LimeSDR: Failed to receive stream
[INFO::Device.py::log_retcode] LimeSDR-CLOSE: Success
[INFO::Device.py::read_receiving_queue] EOF Error: Ending receive thread
[INFO::Device.py::read_device_messages] Exiting read device message thread due to
[DEBUG::Device.py::read_receiving_queue] Exiting read_receive_queue thread.
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread
```

2.7.3 version was installed before and worked successfully

jopohl, 2019-08-20:
Does it help to run ``` LimeUtil --update ``` as suggested?

netdoggy, 2019-08-20:
I have the latest firmware installed. This notification usually means LimeSuite.dll is older than the firmware.
Usually this does not noticeably affect anything.
...
And by the way, I took the relatively fresh LimeSuite.dll from PothosSDR, and it looks like it worked.
LimeSuite URH probably has some kind of problem.

jopohl, 2019-08-20:
Thanks for the hint. Starting with the next release the bundled LimeSuite Version will be 19.04 so that should resolve this issue.

closed 2019-08-20 by jopohl