jopohl/urh #690: macOS: RTLSDR-OPEN (#0): Error Code: -99 (-99)

mcauser, 2019-10-01:
##### Expected Behavior
When I install URH 2.7.5 via pip3, it works fine.
When I use the Mac urh.app, I get an error when trying to record a signal from a RTL-SDR dongle.

##### Actual Behavior
Shows error in the text field at the bottom:
```
RTLSDR-OPEN (#0): Error Code: -99 (-99)
RTLSDR: failed to start rx mode
```

If I run the app manually, the full error message in the Terminal is:
```
Mikes-MBP:~ mike$ /Applications/urh.app/Contents/MacOS/urh
[INFO::SettingsProxy.py::get_receive_buffer_size] Allocate receive buffer with 6.099GB
[INFO::Device.py::start_rx_mode] RTLSDR: Starting RX Mode
usb_claim_interface error -99
[ERROR::Device.py::log_retcode] RTLSDR-OPEN (#0): Error Code: -99 (-99)
[INFO::Device.py::stop_rx_mode] RTLSDR: Stopping RX Mode: Stopped receiving: Stop button clicked
[INFO::Device.py::read_device_messages] Exiting read device message thread due to [Errno 9] Bad file descriptor
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread
[INFO::Device.py::read_receiving_queue] EOF Error: Ending receive thread
[DEBUG::Device.py::read_receiving_queue] Exiting read_receive_queue thread.
```

##### Steps To Reproduce
1. Open urh.app
2. File > Record Signal
3. Set device = RTL-SDR
4. Click refresh icon, Realtek RTL2838UHIDIR (SN: 000...) shown
5. Click Start
6. Error shown in text field.

##### Platform Specifications
 - OS: macOS High Sierra 10.13.6
 - URH version: 2.7.5
 - QT version: 5.13.0
 - Python version: 2.7.16
 - Installed via Github release dmg

jopohl, 2019-10-01:
Works fine on my OS X 10.14 VM. Any chance you can upgrade your OS?

mcauser, 2019-10-01:
I'm going to clean install Mojave soon, so I'll retry after that.

jopohl, 2019-10-01:
Ok great. I updated the Readme and recommend using 10.14+ when using the DMG now. I will close here.

closed 2019-10-01 by jopohl