jopohl/urh #695: pluto not detected

eljuligallego, 2019-11-06:
Hi,
I just installed this tool on ubuntu 18.04 like wiki proposes and tried to run the spectrum analyzer with e pluto board I have but the urh not detect it and the board is correctly connected and running with analog iio-osciloscope app.

Should I have to include some additional commands to enable the native support for this board?

In command line output there is no extra information so I can figure out what can I do.

thanks,
J

eljuligallego, 2019-11-06:
I guess my problem is here:
In order to use native device backends, make sure you install the -dev package for your desired SDRs, that is libairspy-dev, libhackrf-dev, librtlsdr-dev , libuhd-dev 
how I should proceed to install the libpolutosdr-dev?
via pip?....

jopohl, 2019-11-06:
```
apt-get install libiio-dev
```
Then reinstall URH or rebuild device extensions in options.

closed 2019-11-06 by jopohl

eljuligallego, 2019-11-06:
Sorry but, tried to rebuild device after install the libiio-dev but no device detected.
also then update the URH with python3 -m pip install --upgrade urh but no success.

jopohl, 2019-11-06:
What does the HealthCheck say?

eljuligallego, 2019-11-07:
airspy -- ERROR: No module named 'urh.dev.native.lib.airspy'
bladerf -- ERROR: No module named 'urh.dev.native.lib.bladerf'
hackrf -- ERROR: No module named 'urh.dev.native.lib.hackrf'
limesdr -- ERROR: No module named 'urh.dev.native.lib.limesdr'
plutosdr -- OK
rtlsdr -- ERROR: No module named 'urh.dev.native.lib.rtlsdr'
sdrplay -- ERROR: No module named 'urh.dev.native.lib.sdrplay'
usrp -- OK
SoundCard -- No module named 'pyaudio'

eljuligallego, 2019-11-07:
when rebuild the native devices:

Skipping native support for airspy
Skipping native support for bladerf
Skipping native support for hackrf
Skipping native support for limesdr
Found iio lib. Will compile with native plutosdr support
Skipping native support for rtlsdr
Found uhd lib. Will compile with native usrp support
Skipping native support for sdrplay
[1/2] Cythonizing /home/julian/.local/lib/python3.6/site-packages/urh/dev/native/lib/plutosdr.pyx
[2/2] Cythonizing /home/julian/.local/lib/python3.6/site-packages/urh/dev/native/lib/usrp.pyx
running build_ext
building 'urh.dev.native.lib.plutosdr' extension
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.6m -c /home/julian/.local/lib/python3.6/site-packages/urh/dev/native/lib/plutosdr.cpp -o /tmp/home/julian/.local/lib/python3.6/site-packages/urh/dev/native/lib/plutosdr.o
creating build
creating build/lib.linux-x86_64-3.6
creating build/lib.linux-x86_64-3.6/urh
creating build/lib.linux-x86_64-3.6/urh/dev
creating build/lib.linux-x86_64-3.6/urh/dev/native
creating build/lib.linux-x86_64-3.6/urh/dev/native/lib
x86_64-linux-gnu-g++ -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /tmp/home/julian/.local/lib/python3.6/site-packages/urh/dev/native/lib/plutosdr.o -liio -o build/lib.linux-x86_64-3.6/urh/dev/native/lib/plutosdr.cpython-36m-x86_64-linux-gnu.so
building 'urh.dev.native.lib.usrp' extension
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.6m -c /home/julian/.local/lib/python3.6/site-packages/urh/dev/native/lib/usrp.c -o /tmp/home/julian/.local/lib/python3.6/site-packages/urh/dev/native/lib/usrp.o
In file included from /usr/include/python3.6m/numpy/ndarraytypes.h:1809:0,
                 from /usr/include/python3.6m/numpy/ndarrayobject.h:18,
                 from /usr/include/python3.6m/numpy/arrayobject.h:4,
                 from /home/julian/.local/lib/python3.6/site-packages/urh/dev/native/lib/usrp.c:624:
/usr/include/python3.6m/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^~~~~~~
x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /tmp/home/julian/.local/lib/python3.6/site-packages/urh/dev/native/lib/usrp.o -luhd -o build/lib.linux-x86_64-3.6/urh/dev/native/lib/usrp.cpython-36m-x86_64-linux-gnu.so
copying build/lib.linux-x86_64-3.6/urh/dev/native/lib/plutosdr.cpython-36m-x86_64-linux-gnu.so -> urh/dev/native/lib
copying build/lib.linux-x86_64-3.6/urh/dev/native/lib/usrp.cpython-36m-x86_64-linux-gnu.so -> urh/dev/native/lib
/home/julian/.local/lib/python3.6/site-packages/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /home/julian/.local/lib/python3.6/site-packages/urh/dev/native/lib/plutosdr.pyx
  tree = Parsing.p_module(s, pxd, full_module_name)
/home/julian/.local/lib/python3.6/site-packages/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /home/julian/.local/lib/python3.6/site-packages/urh/dev/native/lib/usrp.pyx
  tree = Parsing.p_module(s, pxd, full_module_name)
warning: urh/dev/native/lib/usrp.pyx:204:44: Not all members given for struct 'uhd_tune_request_t'

jopohl, 2019-11-07:
Plutosdr is marked as OK. Does it work when you start URH as root? If so, you need to adapt udev rules.

eljuligallego, 2019-11-14:
Hi, the pluto device doesn't work in root mode with sudo or without....
don't know what else to do.

jopohl, 2019-11-14:
Install libiio?

eljuligallego, 2019-11-14:
libiio is installed, I use pluto board for gnuradio and iio-oscilloscope without problems.
just the URH that don't want to detect it and don't let me debug the connection.
There is not option to enable manually the connection or switch between USB or ethernet connection(set ip manually)?.

jopohl, 2019-11-14:
How do you have it attached? Via USB? Normally the driver automagically finds connected devices.

eljuligallego, 2019-11-14:
yes via USB. the IP is 192.168.2.1

jopohl, 2019-11-14:
Just verified it is working on my Arch Linux machine. What is your libiio version? 0.18 works fine.

eljuligallego, 2019-11-14:
0.18 also and pluto fw version tried with 0.28 and 0.30.
running in ubuntu 18.04.

hangeron, 2019-12-22:
Hi all,
I have a question. How is your Pluto detected? 
Mine is listed twice as "Local devices [local:]". One is always visible, second one appears when I plug the device. But on I don't get any data from it, Spectrum Analyzer is empty even when I get PlutoSDR-OPEN(local:):Success. 
I'm on Ubuntu 18.04, Python 3.6.9, Pluto fw v0.31, urh 2.8.1, iio 0.18.
On GNU Radio 3.8, Pluto is working fine.
Any ideas what I'm doing wrong?