jopohl/urh #702: Issue With LimeSDR Mini

nico0481, 2019-12-01:
Hello,

I can not succeed to use URH with LimeSDRMini, as soon as I try to launch spectrum analyzer, I get:
LimeSDR-OPEN (USB 3.0 (LimeSDR Mini), media=USB, module=uLimeSDR, addr=24607:1027): Success
LimeSDR-INIT: Success
LimeSDR-SET_CHANNEL_INDEX to 0: Success
LimeSDR-SET_ANTENNA_INDEX to 2: Success
LimeSDR-SET_FREQUENCY to 433920000.0: Error Code: -1 (-1)
LimeSDR-SET_SAMPLE_RATE to 2000000.0: Success
LimeSDR-SET_BANDWIDTH to 2000000.0: Error Code: -1 (-1)
LimeSDR-SET_RF_GAIN to 0.25: Success
LimeSDR: Current normalized gain is 0.25
LimeSDR: Current antenna is LNA_L
LimeSDR: Current chip temperature is 44.44°C
LimeSDR: Initializing stream...
LimeSDR-Initialize stream: Success
LimeSDR: successfully started rx mode

**************************************
file linked via:

"If your device does not have a -dev package, e.g. LimeSDR, you need to manually create a symlink to the .so, like this:

sudo ln -s /usr/lib/x86_64-linux-gnu/libLimeSuite.so.17.02.2 /usr/lib/x86_64-linux-gnu/libLimeSuite.so"

as required

**************************************
LimeSDR is recognized by URH (LimeSDR in the pick list and second picklist refreshed with "USB 3.0 (LimeSDR Mini), media=USB, module=uLimeSDR, addr=24607:1027"
************************

LimeUtil --find
  * [LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=1D4C2A888548D4]

URH 2.8.1
Debian Stretch

Thank you

Best regrads

jopohl, 2019-12-01:
Getting LimeSDR working is always painful. Here are some thoughts to get you started:

* The most recent version of LimeSuite is [19.04](https://github.com/myriadrf/LimeSuite/releases) whereas you use 17.02.2. Updating it, and rebuilding the device extenions in URH might do the trick already.
* There is also this  ``` LimeUtil --update ```  stuff which you can play around with. The version installed needs to match the libLimeSuite. However, only certain combinations might work. Be careful however, as you can brick your device with this routine (as I did).

nico0481, 2019-12-01:
Hello,
The version of LimeSuite is 19.04 I just forget to change "sudo ln -s /usr/lib/x86_64-linux-gnu/libLimeSuite.so.17.02.2   ..." I have done the change, and rebuild device extentions, but no success. Neither with LimeUtil --update.

thank you

Best regards

jopohl, 2019-12-01:
Does it work in other applications such as GNU Radio or LimeSuiteGUI?

observator-x, 2019-12-03:
I have an issue too with LimeSDR-Mini under Windows 10, Generator -> Send Signal

It is detected in the Device Identifier, but when sending a message it sends an non-modulated carrier once, after that i have to reset the device by pulling it out and push it in again to the USB slot, with other software under Windows and Ubuntu it works fine, and using latest firmware.

nico0481, 2019-12-04:
Hello,

I've done uninstall ( previously intalled with pip) and fresh reinstall from source.
It seems to work fine now in Rx and Tx.

Thank you

Best regards

closed 2019-12-04 by nico0481