jopohl/urh #707: Issue With LimeSDR Mini under Windows

observator-x, 2019-12-04:
##### Platform Specifications
 - OS: Windows 10

I have an issue with LimeSDR-Mini under Windows 10, Generator -> Send Signal

It is detected in the Device Identifier, but when sending a message it sends an non-modulated carrier once, after that i have to reset the device by pulling it out and push it in again to the USB slot, with other software under Windows and Ubuntu it works fine, and using latest firmware.

LimeSDR-Mini works fine with GNU radio, LimeSuiteGUI etc.

jopohl, 2019-12-04:
Does it work with URH under Ubuntu?

observator-x, 2019-12-05:
I will test this weekend with URH under ubuntu, have 2x Lime-SDR Mini so then will test them both.

observator-x, 2019-12-05:
No issues using Ubuntu

jopohl, 2019-12-05:
What version of LimeSuite is installed in your Ubuntu?

observator-x, 2019-12-05:
Version: 19.04.1-ga5b3a10f
Build date: 2019-11-12

jopohl, 2019-12-06:
Did you build that from source? The latest release on Github is 19.04 (https://github.com/myriadrf/LimeSuite/releases) and this version is bundled URH on windows. You might try to compile this 19.04.1 version on windows and place the resulting DLL in the installation directory of URH to see if that makes a difference.

observator-x, 2019-12-06:
Yes i built it from latest commit (wich is not the latest windows release)

jopohl, 2019-12-08:
Is the latest release working for you? If not, it might be a good idea to open an issue at the limesuite repo.

observator-x, 2019-12-08:
I dont have the latest release of limesuite.dll yet, i installed SDRangel because it uses this dll also but unfortunatly it was an older version too, i think to build the dll from source have to compile with visual studio first, which i dont had no time todo yet.

jopohl, 2020-01-06:
Closing here due to inactivity.

closed 2020-01-06 by jopohl