jopohl/urh #712: Recording with hackRF save wrong format

roema, 2020-01-05:
##### Expected Behavior

##### Actual Behavior

Recording from hackRF doesn't work on Linux

##### Steps To Reproduce

1. Go to 'File -> Record Signal'
2. nothing shown in the right window
3. Terminal show's <ValueError: Buffer dtype mismatch, expected 'float' but got 'signed char'>

##### Platform Specifications

 - OS: Ubuntu 18.04
 - URH version: 2.8.1
 - Python version: 3.6.9

jopohl, 2020-01-05:
Can't reproduce - works fine on Arch Linux. How did you install URH? With pip? Please post the complete terminal output and not just a single line.

roema, 2020-01-05:
Ok.
I am running URH from source.
PIP installation works! Thank you!

[INFO::SettingsProxy.py::get_receive_buffer_size] Allocate receive buffer with 8.133GB
[INFO::Device.py::start_rx_mode] HackRF: Starting RX Mode
[INFO::Device.py::log_retcode] HackRF-SETUP (0000000000000000909864c8373320cf): Success
[INFO::Device.py::log_retcode] HackRF-SET_FREQUENCY to 433920000.0: Success
[INFO::Device.py::log_retcode] HackRF-SET_SAMPLE_RATE to 2000000.0: Success
[INFO::Device.py::log_retcode] HackRF-SET_BANDWIDTH to 2000000.0: Success
[INFO::Device.py::log_retcode] HackRF-SET_RF_GAIN to 0: Success
[INFO::Device.py::log_retcode] HackRF-SET_IF_GAIN to 16: Success
[INFO::Device.py::log_retcode] HackRF-SET_BB_GAIN to 14: Success
[INFO::Device.py::log_retcode] HackRF-Start RX MODE: Success
Traceback (most recent call last):
  File "/home/roema/work/urh/src/urh/controller/dialogs/ReceiveDialog.py", line 62, in update_view
    self.scene_manager.show_full_scene()
  File "/home/roema/work/urh/src/urh/ui/painting/SceneManager.py", line 67, in show_full_scene
    self.show_scene_section(0, self.num_samples)
  File "/home/roema/work/urh/src/urh/ui/painting/SceneManager.py", line 51, in show_scene_section
    subpath_ranges=subpath_ranges)
  File "src/urh/cythonext/path_creator.pyx", line 17, in urh.cythonext.path_creator.create_path
ValueError: Buffer dtype mismatch, expected 'float' but got 'signed char'

jopohl, 2020-01-05:
Your Cython extensions are outdated. Run ``` build.py ``` or make a fresh clone of the repository and it will work from source as well.

closed 2020-01-05 by jopohl