jopohl/urh #719: Provide stream of decoded data

joshwapohlmann, 2020-01-21:
##### Motivation
As I understand it, URH provides the tools to go all the way from signal finding/recording to reverse engineering a protocol. After this, I would like to implement a plugin for wireshark to decode the livestream of received data. This part of the workflow is missing and I have to setup a completly new toolchain to receive data over an SDR and decode it properly until I can dump the stream into wirehshark. 

##### Solution
I want to be able to use URH to stream the received and decoded data to wireshark for live analysis of protocols already reverse engineered.

andynoack, 2020-01-21:
Hi, this is actually not the way we designed URH for. However, is the protocol sniffer dialog (or the content that pops up there) what you need? Maybe, we can implement an interface for the NetworkSDR plugin here (see plugins under options).

joshwapohlmann, 2020-01-21:
@andynoack Thanks for answering so quickly. I'll check this out.
So did you never faced the issue that you picked up some waveform and first wanted to see if you can tap into the stream of packets by using existing protocol parser (e.g. in wireshark)?
What if you want to do so, and inject specific packets or signals? This dual use of URH as a middleware of the SDR pipeline/toolchain.

What are your suggestions for such a usecase? Have you established other workflows for similar ocasions?

andynoack, 2020-01-21:
Indeed we discussed this earlier but the point is: URH is dedicated to proprietary protocols where usually no hardware interface (e.g. WLAN Stick) is available. And when there is no hardware, there isn't a wireshark support available in most cases as well. The other way around: If there is wireshark compatible hardware, then URH is probably not the easiest way to go, because of many things that have to be done in software which are normally done in the dedicated network hardware (modulation and stuff).
If it is easy to implement a network stream from the protocol sniffer, however, we can do it. As pointed out above, the target group for this feature will rather be small.

Can you evaluate:
- If writing to a File (from protocol sniffer) and opening it in an external application helps for your use case?
- If the NetworkSDR plugin helps you
- Which format you need: Bits or hex as ASCII vs. Binary

Then we'll see, if we can do it!

joshwapohlmann, 2020-01-21:
Okay, I get the idea. Thanks for the answers. I will close this issue for the time being until I've done my part since I suppose this will take some time.

closed 2020-01-21 by joshwapohlmann