jopohl/urh #725: Refreshing Device ID with HackRF crashes the application

sliiim9001, 2020-02-03:
##### Expected Behavior

Expected the device ID to propagate into the Device Identifier listbox.  
##### Actual Behavior
Clicked refresh for the Device Identifier field, program crashes with no warning/error message.
##### Steps To Reproduce

1. Go to File -> Spectrum Analyzer.
2. Click on 'HackRF' in the Device selector.
3. Click refresh.
4. See error

##### Platform Specifications

 - OS: Windows 10
 - URH version: 2.8.2
 - Python version: 3.7
 - Installed via MSI 'Universal.Radio.Hacker-2.8.2-x64.exe'

jopohl, 2020-02-03:
Please post the output of ``` %TEMP%\urh.log ```

sliiim9001, 2020-02-03:
%TEMP%/urh.log OUTPUT:
Allocate receive buffer with 262.144KB
HackRF: Starting RX Mode
HackRF-SETUP: HACKRF_ERROR_NOT_FOUND (-5)
HackRF: Stopping RX Mode: Stop on error
EOF Error: Ending receive thread
Exiting read device message thread due to 
Closing parent control connection: handle is closed
Exiting read_receive_queue thread.
Exiting read device errors thread
HackRF: Stopping TX Mode: Stop on error

_______________________________________________

Additional notes: The HackRF is recognized by SDRSharp on the same system without error. UHR works for me with a RTL-SDR.

jopohl, 2020-02-04:
Ths line:
>  HackRF-SETUP: HACKRF_ERROR_NOT_FOUND (-5)

Indicates, that the hackrf driver was correctly loaded by URH. USB on Windows is always a mixture of black magic and luck. What you can try:

1. Update the HackRF USB driver with Zadiq.
2. Try attaching the HackRF to another USB port.
3. Make a full uninstall of URH (delete folder after uninstaling) and then install it again.

sliiim9001, 2020-02-04:
I'll perform these actions when I have some time. 

I discovered that I was able to use the HackRF without specifying the device. From the Spectrum Analyzer window, selecting HackRF as the device and pressing Start, URH would start the analyzer and not crash.

jopohl, 2020-02-04:
That is even more weird. Can you refresh the device list after the device was found (i.e. start/stop in Spectrum analyzer)?

sliiim9001, 2020-02-04:
That produces the same outcome for me. Very strange indeed.

krystian530, 2020-02-06:
I had a similar problem.  It turned out that if I have a Razer mouse plugged in, I can't refresh the device id. When i plug out mouse I can refresh it, then plug in again and everything works

jopohl, 2020-02-09:
So it rather looks like a bug in the hackrf driver then. You might consider to open an issue there. I will close this one, because there is not much we can do about it from URH side. Tip: When you only have one HackRF attached there is no need to refresh the device list. It will automatically connect to the first found HackRF when you do not set a device identifier.

closed 2020-02-09 by jopohl