jopohl/urh #726: LimeSDR on Low Frequencies

BSAsecure, 2020-02-12:
##### Expected Behavior
Better control over GAIN for LimeSDR device (not mini).
Please use example of the SDR Console Gain Control. 

##### Actual Behavior
Signal really weak on the low frequencies even if GAIN is maximized. 
##### Steps To Reproduce
1. Go to URH
2. Click on Spectrum Analyzer
3. Refresh device, select sampling rate to 2.5M and Gain to above middle point (also tested maximum) 
4. See attached screenshots that signal that is strong on the Console is really weak on the URH even if Gain is cranked up to maximum.  

##### Screenshots
Screenshot from Console showing LNA Gain and strong signal reception on RX1_L
[image]

Signal on the URH
[image]

Additional information about LimeSuite:
[image]

Version information:
  Library version:      v19.04.0-PothosSDR-2019.06.09-vc14-x64
  Build timestamp:      2019-06-09
  Interface version:    v2019.1.0
  Binary interface:     19.04-1

Please note that on URH there is no full control over all parameters of LimeSDR  Gain.

##### Platform Specifications
 - OS: Windows 10
 - URH version: [e.g. 2.8.2]
 - Installed via installer

Additional Logs:
LimeSDR-OPEN (LimeSDR-USB, media=USB 3.0, module=FX3, serial=XXXXXX, index=0): Success
LimeSDR-INIT: Success
LimeSDR-SET_CHANNEL_INDEX to 0: Success
LimeSDR-SET_ANTENNA_INDEX to 3: Success
LimeSDR-SET_FREQUENCY to 1000000.0: Success
LimeSDR-SET_SAMPLE_RATE to 2000000.0: Success
LimeSDR-SET_BANDWIDTH to 2000000.0: Success
LimeSDR-SET_RF_GAIN to 0.91: Success
LimeSDR: Current normalized gain is 0.90
LimeSDR: Current antenna is LNAW (I was testing with RX1L and RX1W) 
LimeSDR: Current chip temperature is 37.99°C
LimeSDR: Initializing stream...
LimeSDR-Initialize stream: Success
LimeSDR: successfully started rx mode

xixuedafa, 2020-02-19:
LimeSDR mini also does not work below 30 MHz.Mayhe bellow 30MHz  need to use NCO algorithm. Practically it is UP converter and converts 0.1-30MHz above 30MHz with NCO.

xixuedafa, 2020-02-19:
SDRconsole v3 automatically sets NCO.
SDRSharp like this:
https://github.com/GoranRadivojevic/sdrsharp-limesdr/blob/master/SDRSharp.LimeSDR/LimeSuite.dll

BSAsecure, 2020-02-19:
Make sense that besides lack of smarter Gain  settings using  LNA / PGA / TIA there is need to control NCO. I've noticed that LimeSDR's output power on lower frequencies (CW) while using software that does not control NCO is lower then CW sent via SDR Console.

lonram513, 2020-04-19:
Go to URH
Click on Spectrum Analyzer
Refresh device, select sampling rate to 2.5M and Gain to above middle point (also tested maximum)
See attached screenshots that signal that is strong on the Console is really weak on the URH even if Gain is cranked up to maximum.
Screenshots
Screenshot from Console showing LNA Gain and strong signal reception on RX1_L
image

Signal on the URH
image

Additional information about LimeSuite:
image

Version information:
Library version: v19.04.0-PothosSDR-2019.06.09-vc14-x64
Build timestamp: 2019-06-09
Interface version: v2019.1.0
Binary interface: 19.04-1

Please note that on URH there is no full control over all parameters of LimeSDR Gain.

Platform Specifications
OS: Windows 10
URH version: [e.g. 2.8.2]
Installed via installer
Additional Logs:
LimeSDR-OPEN (LimeSDR-USB, media=USB 3.0, module=FX3, serial=XXXXXX, index=0): Success
LimeSDR-INIT: Success
LimeSDR-SET_CHANNEL_INDEX to 0: Success
LimeSDR-SET_ANTENNA_INDEX to 3: Success
LimeSDR-SET_FREQUENCY to 1000000.0: Success
LimeSDR-SET_SAMPLE_RATE to 2000000.0: Success
LimeSDR-SET_BANDWIDTH to 2000000.0: Success
LimeSDR-SET_RF_GAIN to 0.91: Success
LimeSDR: Current normalized gain is 0.90
LimeSDR: Current antenna is LNAW (I was testing with RX1L and RX1W)
LimeSDR: Current chip temperature is 37.99°C
LimeSDR: Initializing stream...
LimeSDR-Initialize stream: Success
LimeSDR: successfully started rx mode

@jopohl jopohl added feature sdr labels on Feb 12
@andynoack andynoack added the help wanted label on Feb 17
@xixuedafa
 
xixuedafa commented on Feb 18
LimeSDR mini also does not work below 30 MHz.Mayhe bellow 30MHz need to use NCO algorithm. Practically it is UP converter and converts 0.1-30MHz above 30MHz with NCO.

@xixuedafa
 
xixuedafa commented on Feb 18
SDRconsole v3 automatically sets NCO.
SDRSharp like this:
https://github.com/GoranRadivojevic/sdrsharp-limesdr/blob/master/SDRSharp.LimeSDR/LimeSuite.dll

@BSAsecure
 
Author
BSAsecure commented on Feb 19
Make sense that besides lack of smarter Gain settings using LNA / PGA / TIA there is need to control NCO. I've noticed that LimeSDR's output power on lower frequencies (CW) while using software that does not control NCO is lower then CW sent via SDR Console.

@jopohl jopohl mentioned this issue on Feb 25
Limesdr Mini TX under 30mhz #729

ChazTuna, 2020-06-25:
FYI I have the same issue with weak signals when using my SDRplay RSPdx at ~300MHz.  For some reason the native support is not quite working right.