jopohl/urh #738: Error: cannot reshape array of size X into shape

All3xJ, 2020-03-05:
##### Expected Behavior

##### Actual Behavior

##### Steps To Reproduce

1. Go to File
2. Click on Import
3. Import a **.wav** file of signal
4. See error

##### Screenshots

##### Platform Specifications

 - OS: MacOS
 - URH version: [2.8.3]
 - Python version: [3.7.4]
 - Installed via [pip and also dmg, but same error]

```
[ERROR::Errors.py::exception] cannot reshape array of size 85766561 into shape (2)
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/urh/controller/MainController.py", line 783, in show_open_dialog
    self.add_files(file_names)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/urh/controller/MainController.py", line 405, in add_files
    self.add_signalfile(filename, group_id, enforce_sample_rate=enforce_sample_rate)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/urh/controller/MainController.py", line 283, in add_signalfile
    signal = Signal(filename, sig_name, sample_rate=sample_rate)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/urh/signalprocessing/Signal.py", line 70, in __init__
    self.__load_complex_file(filename)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/urh/signalprocessing/Signal.py", line 78, in __load_complex_file
    self.iq_array = IQArray.from_file(filename)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/urh/signalprocessing/IQArray.py", line 203, in from_file
    return IQArray(data=np.fromfile(filename, dtype=np.float32))
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/urh/signalprocessing/IQArray.py", line 17, in __init__
    self.__data = self.convert_array_to_iq(data)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/urh/signalprocessing/IQArray.py", line 212, in convert_array_to_iq
    return arr.reshape((-1, 2), order="C")
ValueError: cannot reshape array of size 85766561 into shape (2)
```

jopohl, 2020-03-05:
Can you provide the wav file please?

All3xJ, 2020-03-05:
> […](#)
> -- Diese Nachricht wurde von meinem Android-Gerät mit K-9 Mail gesendet.

https://we.tl/t-eHT4BHvZ0G

jopohl, 2020-03-05:
You need to give the correct suffix. Renaming 2->2.wav works like a charm.

closed 2020-03-05 by jopohl