jopohl/urh #742: HackRF disabled and can't use it

All3xJ, 2020-03-07:
##### Expected Behavior

I should be able to use HackRF with URH.

##### Actual Behavior
It is transparent in settings, so I can't enable it.

##### Steps To Reproduce

1. Go to Preferences
2. Click on Device
3. Can't enable HackRF

##### Screenshots

[image]

##### Platform Specifications

 - OS: [MacOS Catalina]
 - URH version: [2.8.3]
 - Python version: [2.7.16]
 - Installed via [pip]

I also have installed gr-osmosdr via brew.

jopohl, 2020-03-07:
> ![Schermata 2020-03-07 alle 14 05

Why not use the DMG?

closed 2020-03-07 by All3xJ

innerfirebg, 2020-03-11:
I have the same error but in Linux (parrot os). how can i resolve this?

jopohl, 2020-03-11:
https://github.com/jopohl/urh#generic-installation-with-pip-ubuntudebian

innerfirebg, 2020-03-11:
YES, I install it with pip and everything else and i have gnuradio too.... and my hackrf work with other tools

jopohl, 2020-03-11:
libhackrf-dev installed?

innerfirebg, 2020-03-11:
damn, that missing, thanks!