jopohl/urh #746: Spectrum or record produces error loading 'grc_gnuradio' on 2.8.4

meeas, 2020-03-20:
##### Expected Behavior
Successful spectrum analysis or signal capture

##### Actual Behavior
On the latest 2.8.4, anytime I try a spectrum analysis or signal capture, I get the following error, however rolling back to 2.8.3 everything works fine: 

```
Traceback (most recent call last):
  File "/home/meeas/.local/lib/python3.7/site-packages/urh/dev/gr/scripts/hackrf_recv.py", line 15, in <module>
    from grc_gnuradio import blks2 as grc_blks2
ModuleNotFoundError: No module named 'Traceback (most recent call last):
  File "/home/meeas/.local/lib/python3.7/site-packages/urh/dev/gr/scripts/hackrf_recv.py", line 15, in <module>
    from grc_gnuradio import blks2 as grc_blks2
ModuleNotFoundError: No module named 'grc_gnuradio'
```

##### Steps To Reproduce
1. Go to File > Spectrum Analyzer or File > Record Signal
2. Select appropriate device (I tried both RTL-SDR and HackRF One)
3. Start spectrum analysis or record signal
4. See error

##### Platform Specifications
 - OS: Debian Testing
 - URH version: 2.8.4
 - Python version: 3.7.7
 - Installed via pip3

jopohl, 2020-03-20:
You seem to use the GNU Radio backend for both devices. Does it work when you switch to native in Edit -> Options?

dawud, 2020-03-20:
I'm experiencing the same error in Fedora 32 with an AirSpy2 R2 device, for which I don't have native mode available:

```bash
# rpm -q urh
urh-2.8.4-1.fc32.x86_64

# rpm -q gnuradio
gnuradio-3.8.0.0-5.fc32.x86_64
```

The spectrum analyser fails with:

```bash
Traceback (most recent call last):                                                                                     
  File "/usr/lib64/python3.8/site-packages/urh/dev/gr/scripts/airspy_recv.py", line 15, in <module>                                                                                                                                            
    from grc_gnuradio import blks2 as grc_blks2                                                                        
ModuleNotFoundError: No module named 'grc_gnuradio'
```

Looks like the line

```python
from grc_gnuradio import blks2 as grc_blks2
```

is dead code anyway, as it's never used. It's present in quite a few of the scripts under `urh/src/urh/dev/gr/scripts`.

There are also references to [python2 Queue](https://docs.python.org/2/library/queue.html) that need to be updated to [python3 queue](https://docs.python.org/3/library/queue.html).

I tested locally (for my use case, AirSpy R2), and I can get the spectrum analyser to work, I'll submit a patch for review shortly.

dawud, 2020-03-20:
Before I submit any changes, though, I have a few questions @jopohl:

- some of the `.py` files under `urh/src/urh/dev/gr/scripts` have a header indicating they are generated, are this files supposed to be modified by hand or generated?
- there are quite a few python2 references, both in code and the UI, I assume the plan is to remove them all, is it OK if I just make enough changes for the AirSpy2 R2 device to work?

dawud, 2020-03-20:
A follow-up:

Even though the changes described above allow for the spectrum analyser to start working, as soon as I click in the interface to select a frequency, there's an error:

```python
gr-osmosdr v0.1.x (0.1.5git) gnuradio 3.8.0.0
built-in source types: file rtl rtl_tcp uhd hackrf rfspace airspy soapy redpitaya 
Using AirSpy NOS v1.0.0-rc10-3-g7120e77 2018-04-28, samplerates: 2.5M 10M 
  Airspy decim:1  kernel size:47
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOException in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib64/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib64/python3.8/site-packages/urh/dev/gr/scripts/InputHandlerThread.py", line 36, in run
    input_commands = self.read_input().split("\n")
  File "/usr/lib64/python3.8/site-packages/urh/dev/gr/scripts/InputHandlerThread.py", line 31, in read_input
    result = b"".join(result)
TypeError: sequence item 0: expected a bytes-like object, str found
```

jopohl, 2020-03-21:
> I'm experiencing the same error in Fedora 32 with an AirSpy2 R2 device, for which I don't have native mode available

This works perfectly on Fedora to get native support for RTL-SDR
```
yum install rtl-sdr-devel
yum install urh
```

-------

Regarding GNU Radio: It is expected it will not work with GNU Radio versions >= 3.8 as they switched to Python 3 with this release. The GNU Radio backend, however, was developed, when GNU Radio was Python 2 only. It would require some work to port the Backend to Python 3 (and additonally keeping it compatible with Python 2 for older GR versions).

There is actually no need for the GNU Radio Backend anymore, as all supported SDRs have native support as well, which is faster and better integrated into URH. The only remaining use case for the GR Backend is to easily add support for an SDR not included in URH such as XTRX or so. But I don't know, if anyone is even using that.

Having that said, since users seem to rely on the GR backend anyway I will refactor it with Python 3 support in mind. You can view the progress in #748.

dawud, 2020-03-21:
Thanks @jopohl for the outstanding support. AFAIC this issue can be closed, but I'll let @meeas confirm whether native support works for him as well.

PR #748 (2020-03-22): Overhaul GNU Radio backend (ready for gr 3.8)
This PR will refactor the GNU Radio backend and make it ready for Python 3 (GNU Radio 3.8+). It will also fix #746 .

Steps:
- [x] Update core (support Python 2/3)
- [x] Use Process instead of QThread?
- [x] Update all the scripts
- [x] Without ZMQ? Will UDP source/sink work as well?
- [x] Autodetection of GR installation in GUI 
- [x] Update Docs: No pyzmq required anymore
closed 2020-03-22 by jopohl