jopohl/urh #747: Can not edit frequency after HW selection on 2.8.0+

meeas, 2020-03-20:
##### Expected Behavior
Ability to type new frequency on Spectrum Analysis or Signal Capture

##### Actual Behavior
In the Specrum Analyzer and Record Signal winows, after selecting appropriate hardware in drop down box, the UI does not accept keyboard input into the frequency field, however it does accept keyboard before changing the hardware selection.  Confirmed with issue with all versions 2.8.0 - 2.8.4.  I did not try anything < 2.8.0.

##### Steps To Reproduce
1. Go to File > Spectrum Analyzer or File > Record Signal
2. Click on drop down box to choose appropriate hardware
3. Attempt to type new digits in the frequency field

##### Platform Specifications
 - OS: Debian Testing
 - URH version: 2.8.0 - 2.8.4
 - Python version: 3.7.7
 - Installed via pip3
 - Also running URH under Wayland instead of X11, which might have a part to play in the issue.  But everything else works fine.

jopohl, 2020-03-20:
I can't reproduce this. Here is what I did:

* Installed the latest Debian Testing in a VM and ensured it is running Wayland
* Pasted the install commands from the [Readme](https://github.com/jopohl/urh#generic-installation-with-pip-ubuntudebian), that is
```bash
sudo apt update
sudo apt install python3-numpy python3-psutil python3-zmq python3-pyqt5 g++ libpython3-dev python3-pip cython3
sudo apt install libhackrf-dev liblimesuite-dev libbladerf-dev librtlsdr-dev libairspy-dev libuhd-dev libiio-dev
sudo pip3 install urh
```
* Run URH, go to Spectrum Analyzer, select "HackRF" in Combobox and configure Frequency from 433.92MHz to 433.945MHz in the frequency field
* Verified the same in the Record Dialog

Is there anything different in your procedure?

jopohl, 2020-03-24:
I am closing here due to lacking reaction. Just drop a comment if you feel the issue should be reopened.

closed 2020-03-24 by jopohl