jopohl/urh #752: USRP B100 support.

7doctor7, 2020-03-26:
Hi! I'm trying to use USRP B100 with URH and I can't because if I use native backend gives me:

>UHD Warning:
>    Could not locate B100 firmware. Please run:
>     "/usr/lib/x86_64-linux-gnu/uhd/utils/uhd_images_downloader.py"

if I try to use GNURadio backend, I have:

>[DEBUG::SpectrumThread.py::run] Spectrum Thread: Init Process
>[INFO::AbstractBaseThread.py::stop] Kill grc process
>[INFO::AbstractBaseThread.py::stop] FATAL: Could not find a GR compatible Python interpreter. Make sure you have a running GNU Radio installation.
>[DEBUG::SpectrumThread.py::run] Spectrum Thread: Process initialized
>[INFO::AbstractBaseThread.py::init_recv_socket] Initializing receive socket
>[INFO::AbstractBaseThread.py::init_recv_socket] Initialized receive socket
>[DEBUG::SpectrumThread.py::run] Spectrum Thread: Socket initialized
>[DEBUG::SpectrumThread.py::run] Spectrum Thread: Enter main loop

If I try to see device info by UHD library `$ uhd_find_devices` I have:

>[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.HEAD-0-gaea0e2de
>[INFO] [FX2] Loading firmware image: /usr/local/share/uhd/images/usrp_b100_fw.ihx...
>[INFO] [FX2] Firmware loaded"
>
> UHD Device 0
>
>Device Address:
>    serial: EAR13WDB1
>    name: b100
>    type: b100

Maybe I do something wrong?

jopohl, 2020-03-26:
Please give me some context:
1. What OS are you using?
2. How did you install URH? With pip?
3. What GNU Radio python interpreter did you enter in Options?

7doctor7, 2020-03-26:
1. OS: Linux 4.15.0-91-generic #92-Ubuntu x86_64 x86_64 x86_64 GNU/Linux
2. Installed from source
3. [image]

jopohl, 2020-03-26:
What version of Ubuntu? 16.04?

> Please run:
> "/usr/lib/x86_64-linux-gnu/uhd/utils/uhd_images_downloader.py"
Did you do that?

7doctor7, 2020-03-26:
I use `Linux Mint 19.1 Tessa.`
Yes, I successfully download UHD images and can use USRP B100 in the system.
GNU Radio 3.7.11

jopohl, 2020-03-26:
> Yes, I successfully download UHD images and can use USRP B100 in the system.

What is the error message in URH after downloading the images? Native backend _should_ work now.

> GNU Radio 3.7.11

What is the resukt when you run ``` /usr/bin/python2 -c "import gnuradio" ```  in a terminal? Does it produce an error?

7doctor7, 2020-03-26:
I'm downloaded uhd images using `$ uhd_find_devices` in the terminal, I'm haven't uhd downloader in here `/usr/lib/x86_64-linux-gnu/uhd/utils`

`/usr/bin/python2 -c "import gnuradio"` no error

jopohl, 2020-03-26:
Please make a fresh clone of the repo and post the log when you start URH first time from it.

jopohl, 2020-03-26:
Thinking of it, it is very suspicous that it searches in ``` /usr/lib/x86_64-linux-gnu/ ``` but you have it in ``` /usr/local/ ```. How did you install UHD?

7doctor7, 2020-03-26:
I'll try to reinstall it after some 3-5h...
I installed UHD from source, from they official repo branch v3.15.0

7doctor7, 2020-03-26:
I Installed UHD by instructions from here:
[Building_and_installing_UHD_from_source_code](https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux#Building_and_installing_UHD_from_source_code)

jopohl, 2020-03-26:
Ah, well then it makes sense. You could try to install UHD from Mint's repos, as you seem to have mixed up some paths on Mint. (Not sure what is the reason for  ``` /usr/lib/x86_64-linux-gnu/ ``` path but Mint will have a reason for it). This seems to be not really a URH issue but a Mint one. 

Alternatively, you can simply create a symlink to make URH find the libraries, but this is just an idea, no warranty if you blow up your system. ;)

7doctor7, 2020-03-26:
I tried to do clean install URH from source, and that hadn't effect.

So, I don't know is it a good solution, but I solved my issue by copy UHD lib from `/usr/local/bin/` to `/usr/lib/x86_64-linux-gnu/uhd` and run download script from here, and it works now))

Thanks for your support!))

closed 2020-03-26 by 7doctor7

jopohl, 2020-03-26:
пожалуйста! :)