jopohl/urh #758: HackRF Segfault on device discovery

nightlark, 2020-04-09:
##### Expected Behavior
When I go to the spectrum analyzer, record data, pretty much any function that involves selecting an SDR device, I expect the HackRF to show up in the device list when I click the green refresh button (or just show up in the list without needing to click refresh).

##### Actual Behavior
The current behavior when I select HackRF from the device type is that there are no device identifiers in the list, and when I click the green refresh button in an attempt to get it to populate the drop down list, urh crashes with a segmentation fault.

##### Steps To Reproduce
1. Go to 'File->Spectrum Analyzer..." on Windows.
2. Select Device: 'HackRF'
3. No devices appear in the list. Click green refresh button.
4. urh crashes. Launching it from a command prompt indicates that it is because of segfault.

##### Platform Specifications
 - OS: Windows
 - URH version: v2.0.3 and up have this bug. v2.0.2 was the last version that works for me. Pretty sure the segfault was introduced in https://github.com/jopohl/urh/commit/7407c6b3759ee082d9d1c0d1e2067632c96d670b
 - Python version: Doesn't matter, affects all Python versions.
 - Doesn't matter how it was installed (msi/exe installer or pip)

jopohl, 2020-04-09:
I can't reproduce this on a Windows 10 VM. Works fine there.

Just a side note: There is no need to enter/auto-detect a device identifier for a SDR unless you have more than one of the same type attached. If you leave the device ID empty in your case, URH will use the first available HackRF it finds. 

That said, does it work, when you just click the "record" button and leave the device identifier empty?

nightlark, 2020-04-09:
Just clicking the record button without trying to get a device identifier works.

jopohl, 2020-04-09:
Okay, so you can work with the HackRF, that's good! I added a tooltip to the device identifier to state it is not needed when you have only one SDR of a certain type attached.

Since I cannot reproduce this issue I assume this is not a problem with URH but with the configuration of your system. I guess you have some leftovers of previous installations or other tools which interfere with this HackRF functionality. If you care to investigate this further I suggest you write a little C program to test the device discovery function of the HackRF and open an issue in the HackRF repo if the problem also shows up there.

Since there is not much to be done about this from URH side I will close here.

closed 2020-04-09 by jopohl

nightlark, 2020-04-09:
Is https://github.com/mossmann/hackrf the HackRF repo you're referring to?

nightlark, 2020-04-09:
What I'm seeing in a debugger is that the access violation is occurring in the address space of the `urh\dev\native\lib\hackrf.cp37-win_amd64.pyd` module (source at https://github.com/jopohl/urh/blob/master/src/urh/dev/native/lib/hackrf.pyx#L54-L59?). The one odd thing that I noticed when I stepped into the `init()` call is that of the devices it found, it seemed to locate one usb device that I'd guess is the hackrf, but it didn't hit the calls to libusb for getting an ascii string descriptor, so maybe the `get_device_list()` is crashing when it tries to get a serial number that doesn't exist.

jopohl, 2020-04-09:
Yes this is what was referring to: https://github.com/mossmann/hackrf

What just came to my mind:
1. Is it also crashing when you have no HackRF attached?
2. If yes, is it also crashing if you select a different SDR e.g. RTL-SDR an click the button?

jopohl, 2020-04-09:
This is the full code for ``` get_device_list() ```

```cython
    cpdef get_device_list():
        init()
        cdef chackrf.hackrf_device_list_t* device_list = chackrf.hackrf_device_list()

        result = []
        cdef int i
        for i in range(device_list.devicecount):
            serial_number = device_list.serial_numbers[i].decode("UTF-8")
            result.append(serial_number)

        chackrf.hackrf_device_list_free(device_list)
        exit()
        return result
```

All calls that start with ``` chackrf. ``` come directly from the HackRF API, so if a crash is happening there you should report it at https://github.com/mossmann/hackrf. On URH side of this code I do not see much potential for a segfault in this function.

nightlark, 2020-04-09:
It doesn't crash when no HackRF is attached.

The segfault seems to happen after the `chackrf.hackrf_device_list()` call returns from the HackRF API code. The compiled Python code adds some extra stuff like a call to PyList_New for the returned device list, so I don't know if the line the segfault happens on maps to a line after the `chackrf.hackrf_device_list()` call.

On the HackRF API side, the condition that fails is for a serial number > 0 (https://github.com/mossmann/hackrf/blob/master/host/libhackrf/src/hackrf.c#L438). I'm wondering if the serial number being 0 is related to the unit being ordered from KickStarter a long time ago -- sometime after work I'll look at one of the SDR programs that seems to list a serial number and see if there's anything special needed to get it for an older unit (and check if a firmware update changes anything).

flexd, 2020-06-19:
For what it's worth: This happens to me on Ubuntu 19.10