jopohl/urh #769: On Windows 10 UI does not render, executable is running though

MrBambix, 2020-04-23:
##### Expected Behavior
Upon on clicking the shortcut on the desktop the program should open its main window.

##### Actual Behavior
Actually the Main program window is not showing but proces explorer shows the .exe running

##### Steps To Reproduce

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

##### Screenshots

##### Platform Specifications
Windows 10

jopohl, 2020-04-23:
1. Delete the install folder and reinstall the application with the installer.
2. If that not helps, go to the installation folder, open a terminal (cmd) and run urh_debug.exe and post the output.

MrBambix, 2020-04-23:
> 1. Delete the install folder and reinstall the application with the installer.
> 2. If that not helps, go to the installation folder, open a terminal (cmd) and run `urh_debug.exe` and post the output.

Hi 1 i Tried will no avail unfortunately.. building from source btw seems the same issue when building and running from cmd line without installing 

2. running urh_debug.exe from cmdline no output to be seen in de cmd window..

thanks for you help and suggestions really would like this working with my hackrf ;-)

Regards and Greetings,
Leo

jopohl, 2020-04-23:
Just tested latest master in a Windows 10 VM and it works fine there. Maybe this is some "Run as administrator" / "Windows Defender" voodoo?

MrBambix, 2020-04-23:
hi

Tried running as admin for both installation and running for the regular and debug version. No difference really weird this. Reinstalled my whole php toolchain still no difference. This is an interesting problem 😁

Greetings
Leo

Verstuurd vanaf mijn iPhone

jopohl, 2020-04-23:
Are you able to run a simple hello world PyQt5 script like the following?

```python
# https://pythonprogramminglanguage.com/pyqt5-hello-world/
import sys
from PyQt5 import QtCore, QtWidgets
from PyQt5.QtWidgets import QMainWindow, QLabel, QGridLayout, QWidget
from PyQt5.QtCore import QSize    

class HelloWindow(QMainWindow):
    def __init__(self):
        QMainWindow.__init__(self)

        self.setMinimumSize(QSize(640, 480))    
        self.setWindowTitle("Hello world - pythonprogramminglanguage.com") 

        centralWidget = QWidget(self)          
        self.setCentralWidget(centralWidget)   

        gridLayout = QGridLayout(self)     
        centralWidget.setLayout(gridLayout)  

        title = QLabel("Hello World from PyQt", self) 
        title.setAlignment(QtCore.Qt.AlignCenter) 
        gridLayout.addWidget(title, 0, 0)

if __name__ == "__main__":
    app = QtWidgets.QApplication(sys.argv)
    mainWin = HelloWindow()
    mainWin.show()
    sys.exit( app.exec_() )
```

MrBambix, 2020-04-23:
hi

Will try that later tonight and let you know

Greetz
Leo

Verstuurd vanaf mijn iPhone

MrBambix, 2020-04-23:
> Are you able to run a simple hello world PyQt5 script like the following?
>
> ```python
> # https://pythonprogramminglanguage.com/pyqt5-hello-world/
> import sys
> from PyQt5 import QtCore, QtWidgets
> from [...]

Hi,

Nope same problem not showing the window, its getting weirder...

greetings,
Leo

jopohl, 2020-04-23:
Ok, so there seems to be problem with your PyQt installation. Please run this:

```
python -m pip install --upgrade pip
python -m pip install --upgrade pyqt5
```

and see if it helps.

MrBambix, 2020-04-23:
> Ok, so there seems to be problem with your PyQt installation. Please run this:
> 
> ```
> python -m pip install --upgrade pip
> python -m pip install --upgrade pyqt5
> ```
> 
> and see if it helps.

Hi,

No still nothing, i did however still have a php2.7 installatation on my pc which i just removed. Also uninstalled and reinstalled pyqt5 but still nope.. damn. Maybe some stuff from 2.7 stayed behind somewhere?

greetings,
Leo

jopohl, 2020-04-23:
That is probably unrelated, as a Python3 process will not use any libs from Python2 (except there is some weird naming convention on Windows). 

Having said that, I am running out of ideas what could cause this behaviour. On Linux you could strace the application. I  just saw there is also an strace for [cygwin](https://cygwin.com/cygwin-ug-net/strace.html), so you could install cygwin and install that. However, this appears to be a problem with PyQt5 itself as the hello world example is also not working. Therefore, you may be better off asking there for help.

MrBambix, 2020-04-24:
Hi Johannes

Thanks for all your help I will ask in the qt5 forum and post an update here.

Greetings
Leo

Verstuurd vanaf mijn iPhone

MrBambix, 2020-04-24:
Hi Johannes,

Not even qt creator 4.12.0 runs on my machine.. something seriously wrong with anything Qt on this laptop..

 

Regards,

Leo

jopohl, 2020-04-24:
It appears so! Did you get some reply in the Qt forums?

MrBambix, 2020-04-25:
Hi Johannes,

After a great session on the #pyqt IRC channel it was discovered that it is a bug in the QT release currently in URH. When we rolled it back all started working just fine.

 

So now URH is running here with Qt5Core 5.14.10

 

Regards

 

Leo

jopohl, 2020-04-25:
Mhh so PyQt5 5.14.2 is bugged on Windows 10, thats interesting. Is there an official bug report about it?

MrBambix, 2020-04-25:
Hi Johannes

Yes it is. No not yet they didn't have the time to get a full debug running on my laptop. 

They said maybe the will log the issue regardless.

Maybe you can work together with them on the freenode #pyqt IRC channel.

Greetings
Leo

Verstuurd vanaf mijn iPhone

jopohl, 2020-04-27:
I was interested whether this is a general problem on Windows 10 or not. However, in my VM I cannot reproduce this behaviour so I will not exclude PyQt5.12 from URH dependencies in general. I will close here and I am glad your problem is solved!

closed 2020-04-27 by jopohl