jopohl/urh #774: Jam chirp function

khanfar, 2020-05-12:
please add function that active chirp jam so the tx hover backward and forward in specific frequency in spectrum+ , - , center , like if i want to jam 433 mhz , when i active it , the tx move backward and forward in a loop to 422 then go back to 433 then up to 433 for example , so to jam specific frequency , you see tx in spectrum back and forward arround that frequency in same sdr bandwidth

andynoack, 2020-05-13:
Hi, for what purpose do you need chirp jamming? I am not sure whether this should be an official feature of URH...
You can, however, already implement such a behaviour using the generator with FSK, several bits per symbol and an according bit/hex sequence.

jopohl, 2020-05-13:
Additonally, you can enable __modulation profiles__ in Edit -> Options -> Generation. This allows you to configure modulators in Generation tab __per message__ as an alternative way to achieve your goal.

closed 2020-05-13 by andynoack