jopohl/urh #775: Default NRZi decoder configured incorrectly

hllhll, 2020-05-19:
##### Expected Behavior
NRZ-i Decoder should perform NRZ-i (or NRZi) decoding, I.E. A change from x->!x is 1, else 0
 0100011110 should translate into
   110010001
##### Actual Behavior
Instead it just inverts the data (`Invert stock decoding`)

##### Additional info:
It seems that the stock software decoder for NRZi is configured as `invert` and not as `differential encoding`

andynoack, 2020-05-19:
Thanks for the hint, i just fixed it!

closed 2020-05-19 by andynoack

hllhll, 2020-05-21:
I Guess this is also another way to fix :)

While I'm here I must say this is an incredible piece of software, for my current project i'm using virtually ALL OF THE PROGRAM FEATURES! without writing single line of code.

andynoack, 2020-05-21:
Yes, we discussed it internally and found several resources with different NRZ-I definitions. We think, it is the best solution to avoid the name NRZ-I to not confuse anyone.

commit 749cf3d0 (2020-05-19): Change NRZ-I decoding from invert to differential encoding (see e.g. https://en.wikipedia.org/wiki/Non-return-to-zero). Fix #775
commit 1d3981ca (2020-05-19): Remove NRZ-I to prevent mixing with NRZ + Invert, clarify names. (#776)

* Change NRZ-I decoding from invert to differential encoding (see e.g. https://en.wikipedia.org/wiki/Non-return-to-zero). Fix #775

* skip bugged cython version

* Remove "Non Return To Zero Inverted (NRZ-I)" due to contradictory definitions in literature.