jopohl/urh #778: Skip demodulation

ikarus23, 2020-05-25:
Hi,

I've used this great tool to solve a challenge from the https://www.hackasat.com/ CTF. Once again: Thank you so much for this tool! By solving the challenge I noticed a interesting use case.

URH has it all! Demodulation, clock recovery, line encodings, etc. However, what if the source signal is already demodulated? I was not able to find a way to skip demodulation. I really would like the dropdown menue with ASK, FSK and PSK to have another option: "Skip". This would allow me to still use all the awesome features for signal analysis and generation on a already demodulated signal.

jopohl, 2020-05-25:
Thank you very much, I am glad URH is so useful for you!

What do you exactly mean by "the source signal is already demodulated"? Do you have the bits already or you have a rectangular signal?

* If you have the bits already, you can save the bits in a .txt file (must end with .txt), for example, ``` echo "1010101001\n11110000" > /tmp/some_bits.txt ``` and open it in URH to perform Analysis and Generation.
* If you have a rectangular demodulated singal in IQ format shouldn't it be enough to demodulate it as ASK to get to the same result as choosing "Skip"? If not, it would be helpful if you could upload a test signal.

ikarus23, 2020-05-26:
By demodulated I mean demodulated. ;) So I still have a waveform, but the no clock recovery has been done yet. So no ones and zeros thus far, only a somewhat rectangular signal. Have a look a the screenshot:
[image]
And yes, you are right, this looks like I could use ASK as "Skip" just to track the amplitude. However, if I do so, the demodulated waveform looks strange:
[image]
It's hard to use this result to recover the bit. Maybe this is just a bug after all? Or am I doing something wrong?

jopohl, 2020-05-26:
Yeah the demodulated signal does not look so good after all. There are many interruptions. Maybe a moving average filter might help here?

Apart from that, I see that you are loading a wav file which is not the best format to work with.

ikarus23, 2020-05-26:
Well, the moving average filter changes things, but does not make that much better.

Sure, wav files are not preferred by URH but I guessed it worked because the analog display of the signal was correct (cross checked with baudline).

ikarus23, 2020-05-26:
Maybe ASK does not work because there are no complex numbers in the file. It's just a wav. 16 bit floats, single channel. So you can't do a complex to magnitude transformation (which is typically done for ASK).

jopohl, 2020-05-26:
Ah it is single channel, that explains a lot! Yes you are right, in that case the ASK demodulation will make you lose the sign of the amplitude so you cannot distinguish between 1 and 0 anymore.

ikarus23, 2020-05-27:
But since most (all?) demodulated signals are single channel a "Skip" option might add value to URH.

jopohl, 2020-05-28:
Yes I totally agree with you. In fact, I can't think of a useful way to use a single channel wav with URH except from interpreting it as already demodulated data. I implemented this in the ``` demodulated ``` branch. Can you check that out and see if it works for your signal too?

ikarus23, 2020-05-28:
Damn, how are you always this fast with implementing stuff? ;)

I've tried to build the branch but I got an build error.
```
[ 8/11] Cythonizing src/urh/dev/native/lib/hackrf.pyx
warning: src/urh/dev/native/lib/chackrf.pxd:85:28: 'hackrf_device_list' redeclared 
[ 9/11] Cythonizing src/urh/dev/native/lib/limesdr.pyx
[10/11] Cythonizing src/urh/dev/native/lib/rtlsdr.pyx
[11/11] Cythonizing src/urh/dev/native/lib/usrp.pyx
warning: src/urh/dev/native/lib/usrp.pyx:204:44: Not all members given for struct 'uhd_tune_request_t'
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: no commands supplied
```

jopohl, 2020-05-28:
A clean codebase always pays off. ;) 

Ah yeah, there was a bug introduced in Cython a while ago. Does it help when you run ``` pip install cython!=0.29.18 ``` and try to build again? How are you building by the way? Preferred way is to run ``` src/urh/cythonext/build.py ```

ikarus23, 2020-05-28:
I was using cython 0.29.19. Downgrading to 0.29.17 worked. I also had an issue with `pyqt5` when running `python setup.py install` (in a virtual python environment). After installing `pyqt5` with `pip install pyqt5` manually, I was able to build & install URH.

And the result was great! It works perfectly with the wav! One question: do I need to use a wav? Or can I use a raw file as well and select something like "Skip" or "Single Channel"?

jopohl, 2020-05-28:
Yes, you need to use wav as it is the only way to determine whether the data is one or two channel. I am hesistant to add a general "Skip" functionality to URH as it may confuse users (it is already quite packed with features and options). I think having a raw demodulated file is rare in practice anyway and you could always convert such a file to a single channel wav, if neccessary.

I am glad to hear this branch solves your problem! I will merge the branch into master now.

PR #780 (2020-05-28): add support for already demodulated signals
fix #778 

Treat single channel wav files as already demodulated data.
closed 2020-05-28 by jopohl