jopohl/urh #779: Raspberry Pi Installation

woofwoof, 2020-05-26:
Unsure if it is me but I found Raspberry Pi installation not so straightforward. I ended up installing from source. Its working OK now I think. Pip and snapd did not work for me with numerous issues. However, one thing to note is that upon trying to use setup.py I was prompted to install cython? I had to use "sudo su" so that cython was installed across the system and not just user 'pi' then setup tools worked fine. Maybe this was my issue with pip and snapd? So anyhow, it looks like URH is now running ok though I've yet to put it through its paces. Thanks! I was just wondering about mentioning the Raspberry Pi in install instructions?

jopohl, 2020-05-26:
> Pip and snapd did not work for me with numerous issues

What issues? Without a proper log it is hard to diagnose anything.

woofwoof, 2020-05-26:
Hi. What you say is of course true. I didn't take notes about what went wrong though as I was more interested in what went right. I was simply hoping my comments could be of help to those encountering similar issues. If ever I find time to repeat what didn't work I'll let you know. Until that time I'm happy it seems to be AOK at this time.

jopohl, 2020-05-26:
Ok, cool. Just open a new issue if you ever do another installation on a Raspberry Pi. I do not possess a Pi myself, so I cannot test it. Happy it worked in the end!

closed 2020-05-26 by jopohl