jopohl/urh #792: 20.04 UHD b205 environmental variable incorrect

alphafox02, 2020-07-18:
##### Expected Behavior

Load URH and be able find and use the b205 radio
##### Actual Behavior

URH loads and says it could not find path for image: usrp_b200_fw.hex 

Set the environment variable ‘UHD_IMAGES_DIR’ 

Then goes on to say please run 

“/usr/local/lib64/uhd/utils/uhd_images_downloader.py”

On 20.04 this tool is in /usr/lib/uhd/utils 

So I believe UHD is just looking in the wrong place for the files. Gqrx does not have this problem. 

##### Steps To Reproduce
Start URH from the command line 

Try to select usrp in the spec analyzer drop down box and click refresh icon 

Look at console output

##### Platform Specification
 - OS: Lubuntu 20.04 
 - URH version: 2.8.8
 - Python version: 3
 - Installed via pip

jopohl, 2020-07-18:
I guess this is because with pip you are using the precompiled device backends which may have some settings coming from the build environment (centos 6). If thats true, I see 3 ways to solve this:
1. Set the UHD_IMAGES_DIR environment variable before running URH. From reading your text, I assume you already did this and got it running this way.
2. Uninstall URH and force recompilation when reinstalling ``` pip uninstall urh && pip install urh --no-binary urh ```
3. Use the URH [snap](https://snapcraft.io/urh).

jopohl, 2020-07-18:
These messages are coming from the UHD API by the way. There is not much to be done about this from URH side. I am closing here.

closed 2020-07-18 by jopohl

alphafox02, 2020-07-18:
> I guess this is because with pip you are using the precompiled device backends which may have some settings coming from the build environment (centos 6). If thats true, I see 3 ways to solve this: [...]

#2 appears to have worked great. Thank you. I’ll apply this fix in my project.