jopohl/urh #795: Add support for baseband signal

mminnella, 2020-08-03:
I would like to import the already demodulated data (via CSV file) into URH and the analyze it.
Actually I'm able to import this kind of data via CSV and it looks fine in the analog view, but then URH want to apply one of ASK /FSK /PSK before continue with the analysis phase
I've extracted these data via sigrok on the device before they are sent to the radio.
Anyway due the fact tha URH is great for analyzing bitstream it could be also used also for generic bitstream analysis.

Maybe adding a "Baseband" button along "ASK/FSK/PSK" in the modulation drop down list, with the related code could make the trick.

jopohl, 2020-08-03:
Actually this is already supported. You just need to have your signal as a single channel wav file and load it into URH. Cheers!

mminnella, 2020-08-04:
Thank you for your support!

closed 2020-08-04 by mminnella