jopohl/urh #796: SDRPlay: failed to start rx mode - RSP1A - Ubuntu 1804 - URH 2.8.8

waterwin, 2020-08-11:
##### Expected Behavior
Connected SDRplay RSP1A (working in GNUradio 3.7.11 and CubicSDR) on Ubuntu (virtual client) in Windows10 Oracle VirtualBox host.
Want to do a first record of a signal.

##### Actual Behavior
SDRPlay: CONNECTED DEVICES: RSP 255 (19xxxxxx96)
SDRPlay-SET DEVICE NUMBER to 0: FAIL (1)
SDRPlay: Skipping antenna selection for RSP1 device
SDRPlay-Start RX MODE with 
  FREQUENCY=433920000.0
  SAMPLE_RATE=2100000.0
  BANDWIDTH=2100000.0
  GAIN=29
  IF_GAIN=450: HARDWARE ERROR (7)
SDRPlay: failed to start rx mode
##### Steps To Reproduce

Install URH on Ubuntu, connect a previously installed RSP1A via USB and Run

Tried with different IF_Gain =0 but I doubt that is an issue

Tried with setting to GNUradio instead of native
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/urh/dev/gr/scripts/sdrplay_recv.py", line 10, in <module>
    import osmosdr
ImportError: No module named osmosdr

##### Platform Specifications

 - OS: [e.g. Arch Linux] Ubuntu 1804.4
 - URH version: [e.g. 2.4.2] 2.8.8
 - Python version: [e.g. 3.7] 3.6.9
 - Installed via [e.g. pip] pip

jopohl, 2020-08-11:
I do not have a RSP1A so I cannot give support for this specific device. Just tested my RSP 2 on Arch Linux and it works fine. Could you post the complete log when making a capture (the one when starting URH from command line).

waterwin, 2020-08-11:
I checked this again: SoapySDRUtil -probe=sdrplay
And was amazed to find:
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device sdrplay

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=PlutoSDR
  hardware=ADALM-PLUTO
etc

A mix of finding a SDRplay and using the Pluto driver for this? Including stating it has a Tx channel.

So I removed the SoapySDRPlay directory and reinstalled from git.

Now it works
SDRPlay: CONNECTED DEVICES: RSP 255 (1903175996)
SDRPlay-SET DEVICE NUMBER to 0: Success
SDRPlay: Skipping antenna selection for RSP1 device
SDRPlay-Start RX MODE with 
  FREQUENCY=433920000.0
  SAMPLE_RATE=2100000.0
  BANDWIDTH=1000000.0
  GAIN=29
  IF_GAIN=450: Success
SDRPlay: successfully started rx mode
SDRPlay-SET_IF_GAIN to 1,62K: Success
SDRPlay-SET_IF_GAIN to 450: Success
SDRPlay-SET_RF_GAIN to 54: Success

Thanks for your reaction.
Erwin
PE3ES / F4VTQ

closed 2020-08-11 by waterwin