jopohl/urh #797: Adalm Pluto appears to be connected but not receiving signal at 433.92 MHz

waterwin, 2020-08-11:
##### Expected Behavior
Connect Adalm Pluto, start URH, choose spectrum analyzer, see connection messages to Pluto, push button on KlikAanKlikUit transmitter for 433.92 MHz.
See a signal in the frequency spectrum above noise.

##### Actual Behavior
PlutoSDR-OPEN (usb:1.4.5): Success
PlutoSDR-SET_FREQUENCY to 433,92M: Success
PlutoSDR-SET_SAMPLE_RATE to 2,1M: Success
PlutoSDR-SET_BANDWIDTH to 2,1M: Success
PlutoSDR-SET_RF_GAIN to 15: Success
PlutoSDR: Initializing PlutoSDR..
PlutoSDR: successfully started rx mode

With CubicSDR and Pluto I can receive my KlikAanKlikUit at 433.92 With URH although all messages are ok for the Pluto but no signal is received.

##### Steps To Reproduce
See above with expected behaviour

##### Platform Specifications

 - OS: [e.g. Arch Linux] Ubunt 1804.4
 - URH version: [e.g. 2.4.2] 2.8.8
 - Python version: [e.g. 3.7] 3.6
 - Installed via [e.g. pip] pip

jopohl, 2020-08-11:
Did you try to use a higher gain?

waterwin, 2020-08-11:
Yes tried various settings

~$ urh
[INFO::settings.py::get_receive_buffer_size] Allocate receive buffer with 262,144KB
[INFO::Device.py::start_rx_mode] PlutoSDR: Starting RX Mode
[INFO::Device.py::log_retcode] PlutoSDR-OPEN (usb:1.3.5): Success
[INFO::Device.py::log_retcode] PlutoSDR-SET_FREQUENCY to 433,92M: Success
[INFO::Device.py::log_retcode] PlutoSDR-SET_SAMPLE_RATE to 2,1M: Success
[INFO::Device.py::log_retcode] PlutoSDR-SET_BANDWIDTH to 2,1M: Success
[INFO::Device.py::log_retcode] PlutoSDR-SET_RF_GAIN to 71: Success
[INFO::Device.py::log_retcode] PlutoSDR-SET_FREQUENCY to 433,52M: Success
[INFO::Device.py::log_retcode] PlutoSDR-SET_RF_GAIN to -3: Success
[INFO::Device.py::log_retcode] PlutoSDR-SET_RF_GAIN to -2: Success
[INFO::Device.py::log_retcode] PlutoSDR-SET_RF_GAIN to -1: Success
[INFO::Device.py::log_retcode] PlutoSDR-SET_RF_GAIN to 0: Success
[INFO::Device.py::log_retcode] PlutoSDR-SET_RF_GAIN to 1: Success
[INFO::Device.py::log_retcode] PlutoSDR-SET_RF_GAIN to 12: Success
[INFO::Device.py::log_retcode] PlutoSDR-SET_RF_GAIN to 14: Success
[INFO::Device.py::log_retcode] PlutoSDR-SET_RF_GAIN to 16: Success

Screenshot taken at the same time CubicSDR-KlikAanKlikUit-SDRplay-RSP1A
[image]

jopohl, 2020-08-11:
Does this only affect the Spectrum Analyzer, i.e., are you able to capture a signal with PlutoSDR via File -> Record Signal?

waterwin, 2020-08-12:
That is indeed an interesting suggestion. Record Signal works with Adalm Pluto. From small Gain setting as default but very visible for instance on gain 40

PlutoSDR-OPEN (usb:1.3.5): Success

PlutoSDR-SET_FREQUENCY to 433,92M: Success

PlutoSDR-SET_SAMPLE_RATE to 2,1M: Success

PlutoSDR-SET_BANDWIDTH to 1M: Success

PlutoSDR-SET_RF_GAIN to 40: Success

PlutoSDR: Initializing PlutoSDR..

PlutoSDR: successfully started rx mode

PlutoSDR-CLOSE: Success

andynoack, 2020-08-24:
Close due to inactivity

waterwin, 2020-08-25:
You can close it but the original issue is still not solved ...

Adalm Pluto NOT working with Spectrum analyzer though at the same setup the Record Signal function works ok.

andynoack, 2020-08-25:
We can reopen this issue but note that Adalm Pluto works in our setup, so you need to help us to help you. Otherwise we cannot do anything to help you fixing your setup.

waterwin, 2020-08-25:
When you asked I confirmed that Recording works. What else can I do to help find the setting/parameter I need to correct to make it work better? I can only be active when you tell me what to do.

andynoack, 2020-08-25:
At first we need to exclude that your hardware is defective. Please test your hardware on a fresh environment i.e. a new installation, live cd or vm. I suggest using Manjaro linux as we are currently using Manjaro for developing URH.

waterwin, 2020-08-25:
Let me see, I have that same Adalm Pluto running from a variety of Oracle virtual boxes and from direct systems (Ubuntu 1804, Ubuntu 2004, Windows 10). It works in CubicSDR, GNURadio, GQRX, as well as for the Pothos Recorder. What more could I learn when testing it from a fresh install on Manjaro Llinux?
See some of my notes below:
SDR status Windows10
CubicSDR Pluto+RSP1A OK

GNUradio from Pothos 3.7.13.5 Pluto+RSP1A OK

GQRX from Pothos 2.12 
driver=sdrplay,serial=19xxxxxx96,soapy=3 RSP1A OK
driver=plutosdr
usb:2.5.5
driver=plutosdr,hostname=192.168.2.1
not OK for Pluto
device=plutosdr,driver=plutosdr,soapy=3,uri=usb:2.5.5 OK for Pluto as Unknown

device=plutosdr,driver=plutosdr,soapy=4,uri=usb:1.1.5 werkt op VirtualBox Win10

Pothos Flow RSP1A ok {"driver":"sdrplay"}
Pothos Flow Pluto IIO Device Source geen documentatie
Soapy SDR Source met {"driver":"plutosdr"} werkt wel

GNUradio 3.7.13.5? voorzien van 4 SDRplay blokken RSP1A werkt
15-8
Pluto blocks werken ook zie FM.grc uri ip:192.168.2.1

andynoack, 2020-08-25:
The point is that most of these problems come from a messed up installation of python, some dependencies or drivers. Sometimes also the used hardware plays a role.
We want to figure out what is happening over there. Lets just exclude some things.

Another direction: Are you sure that your signal is strong enough? As spectrum analyzer and recording use the same backend, they only differ in illustration. The Illustration or better the difference between spectrum and recording does not depend on the SDR.

waterwin, 2020-08-26:
As we have in rural France a very slow internet connection I just downloaded the Manjaro iso and will start the URH installation in a Oracle Virtual Box following the instructions. I'll be back!

waterwin, 2020-08-27:
I now have URH working in the Manjaro linux version. First from the package manager and after that through the well documented snap install. Pluto works for all parts.
Then I re-tried the Ubuntu setup. Pluto+URH works in that setup completely and well. Without changes from the previous situation.
So the issue is gone, Thank you for your support and patience,

closed 2020-08-27 by waterwin