jopohl/urh #806: AM-FSK (AM/FSK) interpretation and generation

TwinTurbine, 2020-10-10:
Hi!

I'm trying to interpret a signal that is both AM and FSK modulated. Looking at the signal in URH interpreting it as ASK the signal looks good visually but I cannot interpret it as it is still an FSK layer on top of it. I'm a too inexperienced to say that this isn't possible to perform in URH today but I have no idea how to accomplish that.

Unless this feature already exist I suppose that some kind of layered demodulation/modulation functionality URH would be required.

Being quite a novice in this field I guess that the signal otherwise could be AM-demodulated in some other software and then fed into URH. Being that I also would like to generate signals as well that feels quite cumbersome however.

Any feedback regarding this would be greatly appreciated. Thanks!

andynoack, 2020-10-13:
Hi!

ASK+FSK is rather unusual. Are you sure you don't mean QAM (ASK+PSK)? At the moment URH, however, only supports n-{ASK/PSK/FSK} signals but not a combination of those. Supporting modulation combinations, like QAM, comes with illustration and configuration problems where we need a good solution at first, e.g. how to plot the demodulated signal, which value comes first: ASK or PSK?, ... We always have in mind to not destroy the user experience.

Dear community, how many of you are dealing with modulation combinations? Do you have ideas for solving illustration and configuration (without destroying the UX for people that do not need this)?

andynoack, 2020-10-14:
Hey, we discussed this issue internally and came to the conclusion that layered modulation and demodulation would be to much effort for such a special use case. Hardcoding AM-FSK, however, should not be a big deal! We encourage you to fork the project and modify (at least):
- urh.signalprocessing.Modulator 
- urh.cythonext.signal_functions.modulate_c  
- urh.cythonext.signal_functions.afp_demod

accordingly. From our perspective this would be the best solution because this doesn't impact the "standard" user. If you have questions or need support, don't hesitate to contact us. We recommend our slack channel for discussing detailed implementation questions.