jopohl/urh #809: LimeSDR mini stuck on starting RX mode

alphafox02, 2020-10-28:
##### Expected Behavior

With previous versions of LimeSDR mini I had no issues activating the device to receive signal 
##### Actual Behavior
LimeSDR mini stuck at starting RX mode 

##### Steps To Reproduce

Start URH via command line and then attempt to start LimeSDR mini in any mode 

##### Platform Specifications

 - OS: Ubuntu 20.04
 - URH version: 2.8.9
 - Python version: 3.8

closed 2020-10-28 by alphafox02

zhaohuansi, 2021-01-17:
How did you solve this problem?

alphafox02, 2021-01-17:
Hi. I just walked over and plugged in my mini into a computer I have running DragonOS Focal. When I was building this OS I had this problem I mentioned, but I think the developer told me how to overcome by building in such a way that it used the lime suite installed. Short answer yes it does work now, long answer I’ll have to research and remember what I did to actually make it work.

alphafox02, 2021-01-17:
I think I followed this install method opposed to using package version. 

git clone https://github.com/jopohl/urh/
cd urh
python setup.py install

zhaohuansi, 2021-01-17:
Hello, I can only RX but not Tx under windows and linux environment. And the yellow LED keeps flashing,

alphafox02, 2021-01-17:
Hmm I just recorded for a few seconds, saved then opened it up to transmit. It told me tx successful started and there was just a blinking green link. Seems fine here in Linux. Not sure what would keep it from starting the tx.