jopohl/urh #817: windows installers not working

radioman-00, 2020-11-09:
##### Expected Behavior
Windows program installs ok

##### Actual Behavior
error message: The drive or UNC share you selected does not exists or is not accessible.  Please select another.

##### Steps To Reproduce

ran installers for last two versions

##### Screenshots

[image]

##### Platform Specifications

 - OS: Windows 10, 64-bit

jopohl, 2020-11-09:
Did you actually ensure that the drive you are installing to **is** accessible?

radioman-00, 2020-11-10:
Yes

jopohl, 2020-11-10:
Do you have a prior version of URH installed? If so, uninstall it first.

jopohl, 2020-11-14:
Closing due to inactivity.

closed 2020-11-14 by jopohl

radioman-00, 2020-11-15:
The Windows 64 installer does not offer a choice of installation drive or folder.

jopohl, 2020-11-15:
Repeatedly saying "it does not work" without giving any context what you tried is not helping to diagnose the problem. Here are three solutions out of the blue:
* Uninstall any previous version of URH and clean the registry, then run the installer again
* If that not works, run the installer as Admin
* As a last resort, download and install [Python](https://www.python.org/downloads/release/python-390/) and install URH by typing ```pip install urh```  in a terminal

radioman-00, 2020-11-17:
> Repeatedly saying "it does not work" without giving any context what you tried is not helping to diagnose the problem. Here are three solutions out of the blue:
>
> * Uninstall any previous version of [...]

I have said what the problem is.  I have tried all of the above.  I repeat "The Windows 64 installer does not offer a choice of installation drive or folder."  Please run installer and you will understand.
Thanks M0ZEH

jopohl, 2020-11-17:
> Please run installer and you will understand.

I did that right when you posted the issue, that is what the "cannot reproduce" label means.  On a fresh Windows 10 VM with latest updates the installer works fine.