jopohl/urh #819: Preparing wheel metadata ... error

VinLee, 2020-11-23:
##### Expected Behavior

Expect to install sudo python3 -m pip install urh but i had this failure: 

https://pastebin.com/DLXjzSv4

##### Actual Behavior

Install could not be finished.

##### Steps To Reproduce

What i tried https://pastebin.com/7V9pDekg

##### Screenshots

##### Platform Specifications

 - OS: [https://pastebin.com/wxVDPjyL]
 - URH version: [
Using cached https://www.piwheels.org/simple/urh/urh-2.8.9-cp37-cp37m-linux_armv7l.whl (4.8 MB)
]
 - Python version: [
pi@raspberrypi:~ $ python --version Python 2.7.16
pi@raspberrypi:~ $ python3 --version Python 3.7.3 
]
 - Installed via [
sudo python3 -m pip install --upgrade pip
sudo python3 -m pip install urh
]

jopohl, 2020-11-23:
You have an error installing PyQt5 which is an requirement for URH. Here is the relevant part of the log you posted:

```
Collecting pyqt5
  Using cached PyQt5-5.15.1.tar.gz (3.3 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 /usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmp_4evzfbw
         cwd: /tmp/pip-install-toiy0o72/pyqt5
    Complete output (31 lines):
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 126, in prepare_metadata_for_build_wheel
        hook = backend.prepare_metadata_for_build_wheel
    AttributeError: module 'sipbuild.api' has no attribute 'prepare_metadata_for_build_wheel'
 ```

[Here](https://stackoverflow.com/questions/59780302/pip3-install-pyqt5-user-fails) it is suggested that upgrading pip and setuptools might fix the issue. Another option would be to install another version of PyQt5.

VinLee, 2020-11-23:
Hello Mr. Pohl,

as suggested. 

pi@raspberrypi:~ $ sudo apt-get install python3-pyqt5
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-pyqt5 is already the newest version (5.11.3+dfsg-1+rpi1+b3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

For me it seams that it was already the latest version of pyqt5.

jopohl, 2020-11-23:
I see. Is the installation alright? You can check e.g. with
```
python3 -c "from PyQt5.QtCore import QT_VERSION_STR; print(QT_VERSION_STR)"
```
If everything is fine, you can try to install URH again without using the cached PyQt5 version

```
python3 -m pip install --no-cache-dir urh
```

VinLee, 2020-11-23:
pi@raspberrypi:~ $ python3 -c "from PyQt5.QtCore import QT_VERSION_STR; print(QT_VERSION_STR)"
5.11.3
pi@raspberrypi:~ $ python3 setup.py install
python3: can't open file 'setup.py': [Errno 2] No such file or directory

This cmd python3 -m pip install --no-cache-dir urh 
seams to not finish. Or should I leave it on over night?

jopohl, 2020-11-23:
> pi@raspberrypi:~ $ python3 setup.py install
> python3: can't open file 'setup.py': [Errno 2] No such file or directory

Make sure you clone the repo first and then ``` cd ``` into the directory. Nevertheless:

> This cmd python3 -m pip install --no-cache-dir urh seams to not finish. Or should I leave it on over night?

That is actually a good sign. You could speed things up by installing numpy beforehand with ```  apt-get install python3-numpy ``` beforehand but it could take some time to build the Cython Extensions on a Raspberry. As long as there is some action (check with top if your CPU is active) it means it is compiling.

VinLee, 2020-11-24:
Hello, 

after around 2 hours the compiling have been finished. Thanks for your time and support have a great day.
https://pastebin.com/cABt6rLu

closed 2020-11-24 by VinLee