jopohl/urh #822: Cannot load signal in main application - raspberry pi buster

Fabrice70, 2020-12-05:
Hello,
I cannot load signal in the main application.

I have installed urh on a Raspberry pi 3B+ running with Buster.
 - OS: Raspian Buster
 - URH version: latest ?
 - Python version: 3.7.3
 - Installed via pip but then from source after having issue

Spectrum analyzer is working properly. Then I am going to Record signal and recording it but it seems I have an issue here.

Here is the list of events copied from the command line window:
"
pi@raspi3Bext:~ $ urh
libEGL warning: DRI2: failed to authenticate
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
libpng warning: iCCP: known incorrect sRGB profile
[INFO::settings.py::get_receive_buffer_size] Allocate receive buffer with 262,144KB
[INFO::Device.py::start_rx_mode] RTLSDR: Starting RX Mode
Found Rafael Micro R820T tuner
[INFO::Device.py::log_retcode] RTLSDR-OPEN (#0): Success
Exact sample rate is: 1000000.026491 Hz
[INFO::Device.py::log_retcode] RTLSDR-SET_FREQUENCY to 433.92M: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_SAMPLE_RATE to 1M: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_BANDWIDTH to 1M: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_FREQUENCY_CORRECTION to 1: Success
Disabled direct sampling mode
[INFO::Device.py::log_retcode] RTLSDR-SET_DIRECT_SAMPLING_MODE to 0: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_RF_GAIN to 37: Success
[INFO::Device.py::log_retcode] RTLSDR-RESET_BUFFER: Success
/usr/local/lib/python3.7/dist-packages/urh-2.9.0-py3.7-linux-armv7l.egg/urh/ui/painting/FFTSceneManager.py:24: RuntimeWarning: divide by zero encountered in log10
  paths = path_creator.create_path(np.log10(self.plot_data), start, end)
/usr/local/lib/python3.7/dist-packages/urh-2.9.0-py3.7-linux-armv7l.egg/urh/ui/painting/FFTSceneManager.py:29: RuntimeWarning: divide by zero encountered in log10
  peak_path = path_creator.create_path(np.log10(self.peak), start, end)[0]
[INFO::Device.py::log_retcode] RTLSDR-SET_FREQUENCY to 433.833M: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_FREQUENCY to 433.832M: Success
[INFO::Device.py::stop_rx_mode] RTLSDR: Stopping RX Mode: Stopped receiving: Stop button clicked
[DEBUG::RTLSDR.py::shutdown_device] RTLSDR: closing device
[INFO::Device.py::log_retcode] RTLSDR-CLOSE: Success
[INFO::Device.py::read_receiving_queue] EOF Error: Ending receive thread
[INFO::Device.py::read_device_messages] Exiting read device message thread due to
[DEBUG::Device.py::read_receiving_queue] Exiting read_receive_queue thread.
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread
[DEBUG::Device.py::stop_rx_mode] Closing parent control connection: handle is closed
[INFO::Device.py::stop_rx_mode] RTLSDR: Stopping RX Mode: Dialog closed. Killing recording process.
[DEBUG::SendRecvDialog.py::closeEvent] Device stopped successfully.
[DEBUG::SendRecvDialog.py::closeEvent] Cleaning up device
[DEBUG::SendRecvDialog.py::closeEvent] Successfully cleaned up device
libpng warning: iCCP: known incorrect sRGB profile
[INFO::settings.py::get_receive_buffer_size] Allocate receive buffer with 312,445MB
[INFO::Device.py::start_rx_mode] RTLSDR: Starting RX Mode
Found Rafael Micro R820T tuner
[INFO::Device.py::log_retcode] RTLSDR-OPEN (#0): Success
Exact sample rate is: 2000000.052982 Hz
[INFO::Device.py::log_retcode] RTLSDR-SET_FREQUENCY to 433.832M: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_SAMPLE_RATE to 2M: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_BANDWIDTH to 2M: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_FREQUENCY_CORRECTION to 1: Success
Disabled direct sampling mode
[INFO::Device.py::log_retcode] RTLSDR-SET_DIRECT_SAMPLING_MODE to 0: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_RF_GAIN to 37: Success
[INFO::Device.py::log_retcode] RTLSDR-RESET_BUFFER: Success
[INFO::Device.py::stop_rx_mode] RTLSDR: Stopping RX Mode: Stopped receiving: Stop button clicked
[DEBUG::RTLSDR.py::shutdown_device] RTLSDR: closing device
[INFO::Device.py::log_retcode] RTLSDR-CLOSE: Success
[INFO::Device.py::read_receiving_queue] EOF Error: Ending receive thread
[INFO::Device.py::read_device_messages] Exiting read device message thread due to
[DEBUG::Device.py::read_receiving_queue] Exiting read_receive_queue thread.
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[DEBUG::Device.py::stop_rx_mode] Closing parent control connection: handle is closed
[INFO::Device.py::stop_rx_mode] RTLSDR: Stopping RX Mode: Dialog closed. Killing recording process.
[DEBUG::SendRecvDialog.py::closeEvent] Device stopped successfully.
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/urh-2.9.0-py3.7-linux-armv7l.egg/urh/controller/MainController.py", line 726, in on_signals_recorded
    self.add_signalfile(filename, enforce_sample_rate=sample_rate)
  File "/usr/local/lib/python3.7/dist-packages/urh-2.9.0-py3.7-linux-armv7l.egg/urh/controller/MainController.py", line 288, in add_signalfile
    self.add_signal(signal, group_id)
  File "/usr/local/lib/python3.7/dist-packages/urh-2.9.0-py3.7-linux-armv7l.egg/urh/controller/MainController.py", line 303, in add_signal
    signal.auto_detect(detect_modulation=True, detect_noise=False)
  File "/usr/local/lib/python3.7/dist-packages/urh-2.9.0-py3.7-linux-armv7l.egg/urh/signalprocessing/Signal.py", line 423, in auto_detect
    estimated_params = AutoInterpretation.estimate(self.iq_array, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/urh-2.9.0-py3.7-linux-armv7l.egg/urh/ainterpretation/AutoInterpretation.py", line 393, in estimate
    bit_length = get_bit_length_from_plateau_lengths(merged_lengths)
  File "/usr/local/lib/python3.7/dist-packages/urh-2.9.0-py3.7-linux-armv7l.egg/urh/ainterpretation/AutoInterpretation.py", line 325, in get_bit_length_from_plateau_lengths
    histogram = c_auto_interpretation.get_threshold_divisor_histogram(merged_plateau_lengths)
  File "src/urh/cythonext/auto_interpretation.pyx", line 113, in urh.cythonext.auto_interpretation.get_threshold_divisor_histogram
  File "src/urh/cythonext/auto_interpretation.pyx", line 124, in urh.cythonext.auto_interpretation.get_threshold_divisor_histogram
ValueError: Maximum allowed dimension exceeded
/usr/lib/python3.7/multiprocessing/semaphore_tracker.py:144: UserWarning: semaphore_tracker: There appear to be 2 leaked semaphores to clean up at shutdown
  len(cache))
Abandon
"

Could you help me and give me some guidances ?

Thx a lot
Fabrice

jopohl, 2020-12-05:
Uncheck ``` Edit ``` -> ``` Auto detect signals on loading ```

Fabrice70, 2020-12-10:
Thank you. So easy :).
It is working well.
Great job.
Fabrice

closed 2020-12-10 by Fabrice70