jopohl/urh #823: Filters

woofwoof, 2020-12-14:
##### Is your feature request related to a problem?
I'd like to add filters to urh. I don't see such a feature. Unless I missed something crucial.

##### Describe the solution you'd like
I'm thinking a custom decoding may help? Do I need to code that in C though or could I try python?

##### Describe alternatives you've considered
It would be neat to be able to enter (X dimensional) 'matrices' or (1 dimensional) 'strings' as filters.

##### Additional context
In decoding a present signal I see filters being really useful to highlight features. A simple example: if a 'bit' is 1 and is surrounded by 1s make it zero. Could be similar to a simple 'noise' filter?

andynoack, 2020-12-15:
Hey, you can implement an external decoding in every programming language you like. The problem is the startup time of interpreted languages (e.g. earlier python version - should be faster right now) that introduces some delay which you will feel when your decoding is called many times, i.e. for each line.
Aside from that, your example can probably be realized with the "substitution" base module for custom decodings.

woofwoof, 2020-12-15:
Hi. Ok I'll take a look at "substitution", thanks. I'd like to try a more general solution than just that example though. Thinking matrices again whereby different ones can be 'plugged in'. Thank you for taking the time to reply. Understood re python too  - I'll reconsider C. Best.

andynoack, 2020-12-24:
If this issue is not resolved, please contact us.

closed 2020-12-24 by andynoack