jopohl/urh #824: Instabil auf Windows 10 mit HackRF

Robi2003, 2020-12-23:
Ich hoffe, ich darf hier auf Deutsch schreiben ;-)

##### Expected Behavior
Man geht ja davon aus, dass man die Software startet und einfach nutzen kann.

##### Actual Behavior
Die gesamte Software ist sehr instabil. Sie friert ein oder schließt sich einfach von selbst.

##### Steps To Reproduce
- während des analysierens einer Frequenz
- während der Aufnahme einer Frequenz
- oft passiert das Einfrieren oder Schließen / Abstürzen der Software wenn man diese Vorgänge mit den vorhandenen Buttons beenden möchte (Start / Stop / Clear)

andynoack, 2020-12-24:
Für allgemeine Issues ist es besser, wenn man Englisch schreibt, damit auch die internationale Community etwas davon hat!

In diesem Fall klingt es für mich aber eher nach einem individuellen Problem, denn die Hardware sollte das grundsätzlich bewältigen können. Mein Verdacht ist eine beschädigte Installation oder defekte Hardware.
Am einfachsten ist es, zunächst Hardware-Probleme auszuschließen. Dafür bitte Mal die Systemlogs durchsehen und bei keinem Hinweis testen, ob der Fehler auch unter Linux auftritt (Live-Stick, VM oder Dualboot).

andynoack, 2020-12-30:
Wenn es nicht geholfen hat, einfach nochmal melden!

closed 2020-12-30 by andynoack