jopohl/urh #826: Incorrect array allocation causing crashes

I1138, 2020-12-23:
##### Expected Behavior
The program should open the attached file (captured on a rtl-sdr), and successfully attempt to auto detect on it. 

##### Actual Behavior
The program crashes, attached is the traceback from the most recent source (12/23/20 @ 5pm):
```
Traceback (most recent call last):
  File "/home/ian/source/urh/src/urh/controller/widgets/SignalFrame.py", line 780, in on_btn_autodetect_clicked
    success = self.signal.auto_detect(detect_modulation=detect_modulation, detect_noise=detect_noise)
  File "/home/ian/source/urh/src/urh/signalprocessing/Signal.py", line 423, in auto_detect
    estimated_params = AutoInterpretation.estimate(self.iq_array, **kwargs)
  File "/home/ian/source/urh/src/urh/ainterpretation/AutoInterpretation.py", line 393, in estimate
    bit_length = get_bit_length_from_plateau_lengths(merged_lengths)
  File "/home/ian/source/urh/src/urh/ainterpretation/AutoInterpretation.py", line 325, in get_bit_length_from_plateau_lengths
    histogram = c_auto_interpretation.get_threshold_divisor_histogram(merged_plateau_lengths)
  File "src/urh/cythonext/auto_interpretation.pyx", line 113, in urh.cythonext.auto_interpretation.get_threshold_divisor_histogram
  File "src/urh/cythonext/auto_interpretation.pyx", line 124, in urh.cythonext.auto_interpretation.get_threshold_divisor_histogram
ValueError: Maximum allowed dimension exceeded
Aborted (core dumped)
```
and from the python package:
```
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/urh/controller/widgets/SignalFrame.py", line 780, in on_btn_autodetect_clicked
    success = self.signal.auto_detect(detect_modulation=detect_modulation, detect_noise=detect_noise)
  File "/usr/lib/python3.9/site-packages/urh/signalprocessing/Signal.py", line 423, in auto_detect
    estimated_params = AutoInterpretation.estimate(self.iq_array, **kwargs)
  File "/usr/lib/python3.9/site-packages/urh/ainterpretation/AutoInterpretation.py", line 393, in estimate
    bit_length = get_bit_length_from_plateau_lengths(merged_lengths)
  File "/usr/lib/python3.9/site-packages/urh/ainterpretation/AutoInterpretation.py", line 325, in get_bit_length_from_plateau_lengths
    histogram = c_auto_interpretation.get_threshold_divisor_histogram(merged_plateau_lengths)
  File "src/urh/cythonext/auto_interpretation.pyx", line 113, in urh.cythonext.auto_interpretation.get_threshold_divisor_histogram
  File "src/urh/cythonext/auto_interpretation.pyx", line 124, in urh.cythonext.auto_interpretation.get_threshold_divisor_histogram
MemoryError: Unable to allocate 686. TiB for an array with shape (94240172408931,) and data type uint64
Aborted (core dumped)
```

It appears something about this file, and several others that I recorded, causes this to massively overallocate an array.

##### Steps To Reproduce

Start URH
Open attached file
Click on autodetect parameters
see crash output

##### Platform Specifications
Tested on
 - Arch Linux
 - URH version 2.9.0
 - Python version 3.91
 - Installed via pip, source, (arch linux package also, but uninstalled earlier today, so can't verify with this file)

##### Attached example

Should be a longer file containing two different signals, which was giving me trouble. Referenced in examples above.
[weatherLong.complex16s.tar.gz](https://github.com/jopohl/urh/files/5737323/weatherLong.complex16s.tar.gz)

I1138, 2020-12-23:
The above file was recorded by URH using my rtl-sdr in a fairly quite environment, at about 433.9 MHz, and should contain the data from two weather stations, plus whatever other noise might have been going on. Possibly significant, I had to turn the gain up significantly (350) to overcome the dc correction, so that may have had an affect on the file.

I1138, 2020-12-23:
Edited as I managed to overwrite my example file with tar, and upload an empty archive. The new file displays the same effects, but only has one signal and is much shorter.

jopohl, 2020-12-23:
Thank you for reporting this issue and providing a test file making it a breeze to reproduce. I just merged a fix into ```master```. If you want to try it out in your local clone, do not forget to rebuild the Cython extensions with ```  python src/urh/cythonext/build.py ```.

closed 2020-12-23 by jopohl

I1138, 2020-12-29:
Thanks for the quick fix! I can confirm it's now working!

PR #827 (2020-12-23): fix overflow in AutoInterpretation for certain signals
fix #826