jopohl/urh #828: URH Crashes when selecting HackRF

JosephTheTank13, 2020-12-25:
##### Expected Behavior
When selecting the HackRF, then pressing the reload button, you're supposed to see a string of numbers relating to the HackRF.

##### Actual Behavior
When I go to select my HackRF, I click the refresh button but the program just crashes.
##### Steps To Reproduce
1. Open URH. 
2. Go to any window that lets you select a device, such as the spectrum analyzer window.
3. Select "HackRF" from the pop up box.
4. Press the green "reload" button. 
5. The program will hang for a second, then close.

##### Platform Specifications
 - OS: Windows 10
 - URH version: 2.9.0
 - Python version: 3.9
 - Installed via github release page installer

jopohl, 2020-12-25:
Seems to be the same issue as #758

JosephTheTank13, 2020-12-25:
After switching USB ports, it works but only when I do not press the reload button. The program is still very unstable and crashes constantly for me though.

jopohl, 2020-12-25:
Do you have another SDR to test? I suspect the HackRF driver is a bugged on (some) win 10 installations. In my VM it works fine though.

JosephTheTank13, 2020-12-25:
An RTL-SDR works fine with it. Your theory about the HackRF driver being bugged is plausible, as SDR Angle is showing similar problems when using my HackRF. SDR Angle crashes unless I choose the SoapySDR entry for my HackRF.

closed 2021-01-12 by andynoack