jopohl/urh #835: Unable to start urh

zhaohuansi, 2021-01-13:
Initially, it was installed using pip.  Use normal.  After an abnormal shutdown, it cannot be started.  Reinstallation is also useless.
[image]

Traceback (most recent call last):
  File "/usr/local/bin/urh", line 33, in <module>
    sys.exit(load_entry_point('urh==2.9.1', 'console_scripts', 'urh')())
  File "/usr/local/lib/python3.9/dist-packages/urh-2.9.1-py3.9-linux-aarch64.egg/urh/main.py", line 160, in main
    main_window = MainController()
  File "/usr/local/lib/python3.9/dist-packages/urh-2.9.1-py3.9-linux-aarch64.egg/urh/controller/MainController.py", line 46, in __init__
    self.project_manager = ProjectManager(self)
  File "/usr/local/lib/python3.9/dist-packages/urh-2.9.1-py3.9-linux-aarch64.egg/urh/util/ProjectManager.py", line 58, in __init__
    self.reload_field_types()
  File "/usr/local/lib/python3.9/dist-packages/urh-2.9.1-py3.9-linux-aarch64.egg/urh/util/ProjectManager.py", line 97, in reload_field_types
    self.field_types = FieldType.load_from_xml()
  File "/usr/local/lib/python3.9/dist-packages/urh-2.9.1-py3.9-linux-aarch64.egg/urh/signalprocessing/FieldType.py", line 69, in load_from_xml
    e = ET.parse(settings.FIELD_TYPE_SETTINGS).getroot()
  File "/usr/lib/python3.9/xml/etree/ElementTree.py", line 1229, in parse
    tree.parse(source, parser)
  File "/usr/lib/python3.9/xml/etree/ElementTree.py", line 580, in parse
    self._root = parser._parse_whole(source)
xml.etree.ElementTree.ParseError: no element found: line 1, column 0

 - OS: [kali nethunter Linux]
 - URH version: [2.9.1]
 - Python version: [3.9]
 - Installed via [pip3 make]

jopohl, 2021-01-13:
```bash
rm ~/.config/urh/fieldtypes.xml
```
should do the trick.

zhaohuansi, 2021-01-13:
> ```shell
> RM 〜 /.config/urh/fieldtypes.xml
> ```
> 
> 应该可以。

Very ok
[image]

closed 2021-01-13 by zhaohuansi