jopohl/urh #836: Raspberry Buster can't install

fenyvesi, 2021-01-21:
Raspberry Buster 2021-01-11 URH can't install

##### Actual Behavior

The same error with 3 diffrerent installation method:
command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-i1mojk0v/pyqt5/
##### Steps To Reproduce

1. The proposed  standard solution:
       sudo apt-get install python3-numpy python3-psutil python3-zmq python3-pyqt5 g++ libpython3-dev python3-pip
     sudo pip3 install urh
2. Proposed in bug report
   sudo python3 -m pip install urh
3. Proposed in another bug report:
      sudo su
      pip3 install urh
4. See the same error

##### Platform Specifications

 - OS: Raspberry Buster 2021-01-11
 - URH version: ?
 - Python version:  3.7, pip: 18.1

jopohl, 2021-01-21:
Do you have Cython installed? If not ``` apt-get install cython3 ``` and try installing via pip again afterwards.

fenyvesi, 2021-01-22:
Thank you for the advice, but I have the same error:
command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-i1mojk0v/pyqt5/

Steps:
1. apt-get install cython3
2. sudo pip3 install urh, ERROR
3. sudo python3 -m pip install urh: ERROR

jopohl, 2021-01-22:
Just saw that you are using a quite old version of pip (18.1). Please update it with
```bash
sudo python3 -m pip install --upgrade pip
```
and hopefully have a successful URH installation afterwards. :)

fenyvesi, 2021-01-22:
I'm afraid something is missing yet.

Steps:
1. sudo python3 -m pip install --upgrade pip
2. sudo pip3 install urh
3.  ERROR:

 ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 /usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmp5ijey_o2
         cwd: /tmp/pip-install-6pi2ozx8/pyqt5_22754503f8e543c5823af889cc3b8ce5
    Complete output (31 lines):
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 126, in prepare_metadata_for_build_wheel
        hook = backend.prepare_metadata_for_build_wheel
    AttributeError: module 'sipbuild.api' has no attribute 'prepare_metadata_for_build_wheel'
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
        main()
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
        json_out['return_val'] = hook(**hook_input['kwargs'])
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 131, in prepare_metadata_for_build_wheel
        config_settings)
      File "/usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py", line 159, in _get_wheel_metadata_from_wheel
        whl_basename = backend.build_wheel(metadata_directory, config_settings)
      File "/tmp/pip-build-env-5m4dlxqj/overlay/lib/python3.7/site-packages/sipbuild/api.py", line 51, in build_wheel
        project = AbstractProject.bootstrap('pep517')
      File "/tmp/pip-build-env-5m4dlxqj/overlay/lib/python3.7/site-packages/sipbuild/abstract_project.py", line 83, in bootstrap
        project.setup(pyproject, tool, tool_description)
      File "/tmp/pip-build-env-5m4dlxqj/overlay/lib/python3.7/site-packages/sipbuild/project.py", line 479, in setup
        self.apply_user_defaults(tool)
      File "project.py", line 63, in apply_user_defaults
        super().apply_user_defaults(tool)
      File "/tmp/pip-build-env-5m4dlxqj/overlay/lib/python3.7/site-packages/pyqtbuild/project.py", line 79, in apply_user_defaults
        super().apply_user_defaults(tool)
      File "/tmp/pip-build-env-5m4dlxqj/overlay/lib/python3.7/site-packages/sipbuild/project.py", line 225, in apply_user_defaults
        self.builder.apply_user_defaults(tool)
      File "/tmp/pip-build-env-5m4dlxqj/overlay/lib/python3.7/site-packages/pyqtbuild/builder.py", line 67, in apply_user_defaults
        "specify a working qmake or add it to PATH")
    sipbuild.pyproject.PyProjectOptionException
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.7/dist-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmp5ijey_o2 Check the logs for full command output.

jopohl, 2021-01-22:
Does it work like this?
```bash
git clone https://github.com/jopohl/urh
cd urh
sudo python3 setup.py install
```

fenyvesi, 2021-01-22:
It works. I don't know the reason why the "virgin" Raspberry Buster did not work.
I had several warnings about integer comparison with different signedness.

Thank you for your patience.

jopohl, 2021-01-22:
From what I can tell is that pip tries to install PyQt5 from PyPi which seems to be bugged on the Raspberry Buster when looking at the log you posted. When you install URH with ``` setup.py install ``` it realizes that you have a working version of PyQt5 installed from the repos already and skips this step.

In any case I am glad it works for you now. :)

closed 2021-01-22 by jopohl

fenyvesi, 2021-01-22:
Thanks once more.

fenyvesi, 2021-01-22:
Sorry to be here again.

It is true that URH starts that i tried in the morning but the RTL-SDR feature is disbaled I can't change.

Steps:
1. I see the dongle with lsusb.
2. I made udev rules: in file /etc/udev/rules.d/10-rtl-sdr.rules with the RTL-SDR content.
3. sudo udevadm control --reload-rules, reattach the dongle
4. URH can't see the dongle
5. Reboot
6. The same problem.

jopohl, 2021-01-22:
Is the native backend greyed out under Options -> Device? If so, please install librtlsdr-dev and rebuild the device extensions from within Options -> Device.

fenyvesi, 2021-01-22:
Just a strange coincidence. I am from Hungary. The name URH ( Ultra Rövid Hullám) means: Ultra Short Wave. (The FM spectrum). We had a punk band with this name also.

jopohl, 2021-01-22:
Haha seems like we chose just the right name :D Took us some thinkering indeed back in the days

fenyvesi, 2021-01-22:
I am not sure that I made it well. 
1. In the Options/Device 
           /Native Options 
            /Library Directories wrote the propsed name:  librtlsdr-dev 
2. Rebuild
3. Restart URH.
4. No RTL-SDR, it is greyed out.

jopohl, 2021-01-22:
Nono librtlsdr-dev is a package, so it should be installed with ``` sudo apt-get install librtlsdr-dev ```, sorry for the confusion.

fenyvesi, 2021-01-22:
1. sudo apt-get install librtlsdr-dev
2. URH, RTL-SDR greyed out.
3. Restart
4. The same.

In the meantime I saw that in the terminal window the URH is started from there is a warning message: libEGL warning: DRI2: failed to authenticate.

jopohl, 2021-01-22:
Did you hit the rebuild button after installing librtsdr-dev? If it not works, just reinstall URH. Since you have librtlsdr-dev now it should work fine.

fenyvesi, 2021-01-22:
No.
I rebuilt from URH just now. It says that rebuilt 1 device extension but the result is the same. Greyed out after restarting.

I am trying to reinstall with your proposed solution:
sudo python3 setup.py install

jopohl, 2021-01-22:
If it rebuilt the extension it properly has no permission to write it to the installed location. Running URH as sudo and then rebuilding should solve this.

fenyvesi, 2021-01-22:
I just started installation again. Some minutes yet.

fenyvesi, 2021-01-22:
HURRAY, IT WORKS!

Sorry for robbing your time.

George

jopohl, 2021-01-22:
No worries, enjoy sniffing the wireless spectrum.