jopohl/urh #840: Modifying CRC data bits range crashes URH

Adminiuga, 2021-02-15:
##### Expected Behavior

Being able to modify the range of data to checksum

##### Actual Behavior
With URH 2.9.1 installed from pypi on Debian unstable (Python 3.9.1 venv), attempt to modify the range of data to checksum throws an exception and URH crashes.
##### Steps To Reproduce

1. Open URH
2. Load Project
3. Switch to "Analysis" tab
4. In "Labels for message" box, scroll down to "checksum" label and right-click -> Edit
5. Under "advanced settings" -> checksum -> Configure data ranges for CRC" trying to modify the range or deleting it, causes URH to crash. Current settings are `start: 164` and `end: 184` but the current messages are shorter than this

Exception:
```
Traceback (most recent call last):
  File "/home/lex/.venv/sdr/lib/python3.9/site-packages/urh/ui/delegates/SpinBoxDelegate.py", line 32, in setModelData
    model.setData(index, editor.value(), Qt.EditRole)
  File "/home/lex/.venv/sdr/lib/python3.9/site-packages/urh/controller/widgets/ChecksumWidget.py", line 86, in setData
    data_range[0] = converted_index
TypeError: 'tuple' object does not support item assignment
Aborted
```

##### Screenshots

[image]

##### Platform Specifications

 - OS: Debian Linux BullsEye (testing)
 - URH version: 2.9.1
 - Python version: 3.9.1+
 - Installed via Pip into a python virtual environment

Adminiuga, 2021-02-15:
Adding a second range of data to CRC, produces this result in `project.xml`
```
<checksum_label name="checksum" start="80" end="96" color_index="2" apply_decoding="True" show="2" display_format_index="1" display_bit_order_index="0" display_endianness="big" fuzz_me="2" fuzz_values="" auto_created="True" data_ranges="[(164. 184), [21, 80]]" category="generic">
```

So the 1st range is a tuple and UI probably tries to modify it. Manually modifying the project XML worked for me so perhaps could be used as a work around

andynoack, 2021-02-15:
I have tried, but I cannot reproduce a crash :/ Maybe Johannes can see what has probably gone wrong there...

Adminiuga, 2021-02-15:
While I was able to reproduce the issue reliably, prior the application of the work around, there could be some steps I performed in advanced, before getting into that step. I'm just starting with URH and very likely pressed the wrong buttons :)
The top thing that comes to my mind, originally the data frames where analyzed with NRZ line code which got the frame length about 184 bits, but later changed it to "Manchester" coding which reduced the frame length, but the auto detected "checksum" field still had the older values, outside the bounds of the current data frame length.

Adminiuga, 2021-02-15:
i'll try in the evening to reproduce with the same signal file and a new project.

Adminiuga, 2021-02-15:
> The top thing that comes to my mind,

Another step that I made: after changing the line coding, I've changed the "checksum" field position. I don't recall the exact data, but I do remember that frame was about 96bits long, but the checksum field position was using the old values, the values guessed with the NRZ line coding, which were outside of the frame range after switching to manchester.

Adminiuga, 2021-02-16:
Ok, I'm attaching [ook1-1.complex.gz](https://github.com/jopohl/urh/files/5985059/ook1-1.complex.gz) file I'm using to reproduce the issue. Unpack and copy the same file under two different names, to use as the source signal for analysis.
1. start URH in non-project mode
2. Open the two files for analysis
3. Change signal params for both files to:
  a. Noise: 0.006
  b. Samples per symbol: 610
  c. Modulation: ASK
[image]
4. Switch to Analysis tab and click "Analyze Protocol" button
5. Should get two fields: `preamble` and `synchronization`, synchronization uses the whole data frame
[image]
6. Change `synchronization` field  end at 22
[image]

7. Run analysis again and it should get the "checksum" field
[image]
8. Edit checksum fields -- for some reason now it shows empty range under the advanced settings -> configure data ranges for CRC
[image]

9. edit the "configure data ranges for CRC", when the `start` field gets focus, it auto populates with `46`. After that, when moving cursor/focus to end field, it crashes with:
[image]

closed 2021-02-16 by jopohl

andynoack, 2021-02-16:
Thanks for beeing so verbose with your error messages!

PR #841 (2021-02-16): Use list for checksum data_ranges
The auto created "checksum" field/label is initialized with a list with a single tuple for data ranges, which causes URH to crash if the auto guessed range is modified at any point later via CheckSum widget. 
The `__data_ranges` should be `list[list[int, int]]`

Fixes #840